opentx/opentx #4016: Lua telemetry scripts run/background behavior

projectkk2glider, 2016-11-11:
Telemetry scripts run/background function is a bit of a mess. This was last changed in https://github.com/opentx/opentx/commit/8c9e5687f137e6f0a50627a96d7f1d8e8cdc83e5

That change is in 2.1 as well as 2.2 branch. So both versions should work the same. Here is the gist:
- the Lua docs are not in sync with actual behavior https://github.com/opentx/lua-reference-guide/blob/master/telemetry.md
- the `background()` function is called in every loop regardless of the visibility of the telemetry script/screen.
- the `run()` function is obviously only called if the scripts screen is currently active/visible.

This change was first introduced in the first release of OpenTX 2.1.0 and surprisingly nobody noticed until now. To be fair it would be cleaner if we would keep the old behavior (like it is documented), but the fix will be tough. Currently the background functions of all scripts are executed while the LCD is being refreshed by DMA. For any active script we then have to wait for LCD DMA to finish before allowing it to run (because it might change the display buffer).

Not sure which is the best solution for this:
- change docs and keep current solution
- or fix the current behavior to adhere to the docs

Both options could break/affect the existing scripts

lshems, 2016-11-13:
I certainly noticed, and Made all my scripts compatibele with Both Systems by checking the opentx Version.

I would prefer not to change it, as it makes more sense as it is now for most people.

Performance / memory wise, things should run with and without LCD active, so I don't see limitations.

Thanks

pau33, 2016-11-16:
I would agree to Ishems. When I do background scripting, for example, do Sensor-calcluations with every new point, this should work regardlessly of of LCD active/inactive. So I would change the docs.

Thank You

projectkk2glider, 2016-11-16:
It seems that we all agree on the doc correction.

projectkk2glider, 2016-11-16:
Docs updated in https://github.com/opentx/lua-reference-guide/commit/15ac5d6d24c7e571482db228da33cfe93f0c197b

closed 2016-11-16 by projectkk2glider

lshems, 2016-11-29:
Ok, not there yet? there is this open question I have in post which touches also this subject. https://github.com/opentx/opentx/issues/3377

Since we do now explicitly run the background AND the run function, it would make sense to call them alternatively between the other radio processing slices when the LCD is active. This would allow the scripts to run always correct, and prevent a background script being killed because the run script has become active and is called in the same cycle.

if this analysis is correct, ...

projectkk2glider, 2016-11-30:
>Ok, not there yet? 

Where? Have you looked at https://opentx.gitbooks.io/opentx-lua-reference-guide/content/telemetry.html

> Since we do now explicitly run the background AND the run function, it would make sense to call them alternatively between the other radio processing slices when the LCD is active. This would allow [...]

Not sure what are you trying to say. It is very simple:
 * `background` is called every cycle. You are not allowed to write to LCD here.
 * then some time later `run` is called (if your script is the one active/visible). You do your drawing here. No need to call background in the run function, like it was in OTX 2.0.

lshems, 2016-12-01:
This means that with LCD on, there is More Chance of the script being killed, because both are run in one cycle.

projectkk2glider, 2016-12-01:
No, the execution of `background` and `run` is limited individually, so each one gets the same amount of run time.