opentx/opentx #4021: SF to test the failsafe set by the radio

fraternl, 2016-11-13:
With X receivers it is possible to set the failsafe in the radio.
The only way to test this is by turning off the module.

Doing this in-flight is not something I would dare. Even on the ground this needs to be done with care.

It would give me a lot of peace of mind if I could check the failsafe at a safe height.

There are also stabilizers that can turn a plane right-side up and keep the plane level.

RealTadango, 2016-11-16:
You could also test the deisired flight mode without resorting to failsafe?

fraternl, 2016-11-16:
I have no idea what you mean by that...
I want to know what my plane does when it enters failsafe.
In my radio I can set each channel to a certain value or as hold....  these values needs to be sent to the receiver....

fraternl, 2016-11-16:
@projectkk2glider I think this one would need an EEPROM change as well...

Waldo-P-Barnstormer, 2016-11-16:
See this post to see how I've been using a Flight Mode to both set and check my failsafe settings.
https://www.rcgroups.com/forums/showpost.php?p=35973263&postcount=25098
Works good, is safe to use, and should be accurate, 
Jim

fraternl, 2016-11-17:
Hi Waldo....
Thanks for this post and your recipe does get me to where I want to go.

I will be implementing it in my next model.

3djc, 2016-11-17:
Since an alternate, less dangerous, solution is available, closing this

closed 2016-11-17 by 3djc

RealTadango, 2016-11-17:
But that solution will not support no pulses failsafe modes ect.

Another solution is using a XJT module and add a switch in the power line :) That is faster then rebooting OpenTX....

kilrah, 2016-11-17:
>But that solution will not support no pulses failsafe modes ect.

Nothing we could do can simulate those apart from cutting power to the module.

RealTadango, 2016-11-17:
True but turning the module of through a SF would be a solution. The reason for this featue is a very short recovery time.

kilrah, 2016-11-17:
That was actually part of the "dangerous module functions" at some point, but was removed to make room for another function. 
I believe any new SF is a major task now since the size of the function type needs to change.