opentx/opentx #4023: Distorted sound starts after warning screen and stops when the radio is switched off

fraternl, 2016-11-14:
https://goo.gl/photos/bmXqm1AYGHdg5Zjr9

It happened to me once when I was on 2.1, but now it happened again.
After turning on the radio it gives a constant stuttering sound as if the processor is too busy to handle the sound.
It stops when I reset the radio and create a new model and play a sound.
If I load an eeprom of 2 weeks ago it comes back (then this problem didn't occur).
When I reformatted the SD-Card and copied all files back the problem stayed.
I then swapped the SD-Card with my Horus. I then had still the same problem

When I put my Taranis SD-Card back, all distortion was gone and it didn't make a diffeence which firmware was running (N359 or self-compiled) nor made it a difference if I loaded the eeprom of 2 weeks ago of that of last Saturday.

projectkk2glider, 2016-11-14:
* the issue is always:
 * either present right from the radio start (after the warnings are cleared)
 * or never happens
* the sounds before the main display (warning checks and startup sound) and after shutdown (shutdown sound) are always clear.

Basically only happens when main gui thread is fully running and not every time (very rarely).

fraternl, 2016-11-17:
During all this testing and taking SD-Card out and putting it back in I always kept the battery attached. My Taranis has a power light and I'm sure it was turned off everytime. This power light goes on for instance when I attach it in power-off mode to my computer using the USB-cable.

With retrospect I can remember that the battery got loose one time when switching the SD-Card. I do not know if it was the occasion that the sound became normal. When this happens again I will be testing that explicitly.

closed 2016-11-27 by fraternl