opentx/opentx #4029: Horus Futaba trainer plug does not work, Stops SD card when used!

RealTadango, 2016-11-16:
I just did a pull and build the firmware so i am on the latest code.

The problem is that a PPM signal works great in the jack plug but not in the Futaba plug. When i put the same PPM signal in the Futaba trainer plug it is not accepted AND the SD card stops working also. So... that's not good :)

I need the Futaba plug because of the power output.

projectkk2glider, 2016-11-16:
>I need the Futaba plug because of the power output.

I know that at least beta Hours units did not have any power supply signal on Futaba trainer plug. Did you measure anything?

projectkk2glider, 2016-11-16:
One bug found, `TRAINER_DETECT_GPIO_PIN` on Hours is not initialized.

projectkk2glider, 2016-11-16:
Missing initialization fixed in 4eac2e58a0f77

RealTadango, 2016-11-16:
Power output is working on the plug. Receiver works perfectly.

RealTadango, 2016-11-16:
Change has no effect. No trainer signal detected and SD card still stops

projectkk2glider, 2016-11-16:
I assumed this was not the only problem.

RealTadango, 2016-11-16:
PPM Output in slave mode is working on pin 3.

bsongis, 2016-11-17:
Does it work on FrSky firmware?

RealTadango, 2016-11-17:
Good question. I will try when I get home.

RealTadango, 2016-11-17:
FrTX is not working also :( Sd card seems to keep on working though

RealTadango, 2016-11-17:
Mmh, i can see some activity on the PPM in line when browsing the SD... that's not good. I have a look inside.

RealTadango, 2016-11-17:
3 issues: 
 - Futaba plug is only meant as PPM output, not input
 - PPM Input pin is not connected
 - There is no plug detection

So to get it to work:
Connect PPM in pin (3) to PPM out (6) on the Futaba connector. Connect Pin 1 to the transistor output on the SD PCB.

SD card stopping is a power spike on the SD PCB.

closed 2016-11-17 by RealTadango

bsongis, 2016-11-17:
Thanks for this report!

Casichispea, 2017-09-22:
it doesn´t work. I have conected a receiver´s ppm signal to PPM out (6) and nothing happens. PIN 1 is not connected wiht the receiver, it makes no sense to connect it to the output on SD PCB, or?