opentx/opentx #4031: Telemetry Sensor RSSI tells under rpm - dezibel and many other wrong audiofiles

Githubbachelor, 2016-11-17:
@ Gilrah
Hallo If you want to hear the Dezibel from RSSI you have to choose "rmp" in the editor for DEZIBEL.
If you choose dB you hear milliwatt, for W is ok come Watt, for mAh come only Zero, for g you hear rpm, for ° you hear g, for ml you hear degree °, for fOz you hear Zero, for I--I you hear milliliter, for -Gl you hear UNZE.
Rest of UNITS for EDIT UNITS are right connected to audiofile.

[image]
[image]
[image]
[image]

3djc, 2016-11-17:
what audio files are you using ?

Githubbachelor, 2016-11-17:
I use the audiofiles for DE System

3djc, 2016-11-17:
From http://downloads.open-tx.org/2.2/nightly/sdcard/ ?

Githubbachelor, 2016-11-17:
yes

Githubbachelor, 2016-11-17:
can you open it I make a mistake in Compouter

Githubbachelor, 2016-11-17:
I mean reopen it

3djc, 2016-11-17:
I cannot reproduce, works fine on my radio

3djc, 2016-11-17:
it plays file /SOUNDS/de/SYSTEM/db0.wav, from sdcard. Can you check on your computer that this says db ?

Githubbachelor, 2016-11-17:
What can be the problem?

Githubbachelor, 2016-11-17:
All in Sounds/de System is 100% OK. All Files are working with exactly names

Githubbachelor, 2016-11-17:
If you go in Telemetry and say find all sensors start, it find all. For example RSSI - but show RPM. But tell DB for rpm. If you change in editor to Db it tells milliwatt

3djc, 2016-11-17:
wait, that's a completely different issue

Githubbachelor, 2016-11-17:
later in display you see RSSI 91rpm and tell DB

Githubbachelor, 2016-11-17:
if you take db for RSSI you see in Display later RSSI 91dB and it tell milliwatt

Githubbachelor, 2016-11-17:
But the soundfile dB0, rpm0, aso is ok

Githubbachelor, 2016-11-17:
must be a route mistake to the files or wrong DESIGNATION, I think so

Githubbachelor, 2016-11-17:
Where can I give my idea? To you? For example to make the possibility for Moving/copy lines in the Sensor menu. For example to move RSSI line under the A1 line....

3djc, 2016-11-17:
German is missing mW unit, which shifts db, rpm, g, ...

closed 2016-11-17 by bsongis

Githubbachelor, 2016-11-17:
Why do you close that? That was a confirmed bug!
Greetings Michael

schwabe, 2016-11-17:
Because he commited a fix.

3djc, 2016-11-18:
This will be fixed in next available nightly

kilrah, 2016-11-19:
> Where can I give my idea? To you?  

You create a new issue with the suggestion.

PR #4032 (2016-11-17): Fix german units
Telemetry unit translation seem of (not sure why that would be the case)
This fixes #4031