opentx/opentx #4036: Horus hang up in Black Screen and  Blue Screen

Githubbachelor, 2016-11-17:
If you choose Blue Background Screen in Telemetry, and go back in to the model sites to the mixer Site the Screen is blocking and show unclear colored screen, see pictures.

Also if you scroll through the Model Menus my system block 3 times during preparing a model, this evening. You need a very long Exit on Main switch.

Must be a big bug.

If you choose the Black Background in Telemetry the system hang up in the mixersite. in both problems you have to do a long switch off the main switch to exit.
[image]
[image]
[image]

3djc, 2016-11-18:
Did you copy the whole SD card zip file ? It looks to me like you may have taken only the sounds and may be missing some files in the THEMES folder

bsongis, 2016-11-18:
@3djc do you know if is it reproductible?

3djc, 2016-11-18:
I cannot reproduce this

bsongis, 2016-11-18:
It would be interesting to have the SD contents from @Githubbachelor to be able to reproduce it, we miss some check somewhere

3djc, 2016-11-18:
It looks to me like drawing outside of the screen,, expereienced that a few times during the wizard handler dev. I'm having a look at the dark blue today, it is broken in many places

bsongis, 2016-11-18:
@3djc as soon as you find a way to reproduce this problem of drawing outside the screen please tell-me, I think we should fix it

Githubbachelor, 2016-11-18:
1. Yes I have the actuall Card. You can reproduce the problem with blue screen. 

2. That the system is hanging sometimes, maybe the system is too slow????? My Model Programm is big, with sounds and Infos. 
I try to load the system Sound. After rest of card
[SYSTEM.zip](https://github.com/opentx/opentx/files/599263/SYSTEM.zip)

[SD Karte Anlage Horus letzte version aktuell.zip](https://github.com/opentx/opentx/files/599265/SD.Karte.Anlage.Horus.letzte.version.aktuell.zip)

Githubbachelor, 2016-11-18:
I can answer more later at 5h pm earliest

3djc, 2016-11-18:
Tested all the 14 models, cannot reproduce

Githubbachelor, 2016-11-18:
Ok I will try again in the afternoon. I got a new Horus 2 days ago. The first Horus had a TX damage and was changed now Albert Lacroix.

3djc, 2016-11-18:
can you tell me what model it was ?

Githubbachelor, 2016-11-18:
It was Polizia X8 under Heli. I dont know the Modelnumber

3djc, 2016-11-18:
[image]
???

Githubbachelor, 2016-11-18:
Moment
[SYSTEM.zip](https://github.com/opentx/opentx/files/599311/SYSTEM.zip)
[Archive.zip](https://github.com/opentx/opentx/files/599312/Archive.zip)

try this. Sorry I didnt save that night. That is actual

Githubbachelor, 2016-11-18:
Model Polizia Heli

Githubbachelor, 2016-11-18:
I must go now

3djc, 2016-11-18:
Right, this model segfault with blue

3djc, 2016-11-18:
ok, actually quite difficult to reproduce, but there are issues switching from default to dark blue. Here is an example of the wrong mask been loaded (bellow ch3, this is not the right icon)

[image]

3djc, 2016-11-18:
And this is when starting straight in blue

[image]

3djc, 2016-11-18:
@bsongis , I may need some help here:

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc8899700 (LWP 7564)]
__memcpy_sse2_unaligned () at ../sysdeps/x86_64/multiarch/memcpy-sse2-unaligned.S:33
33	../sysdeps/x86_64/multiarch/memcpy-sse2-unaligned.S: Aucun fichier ou dossier de ce type.

3djc, 2016-11-18:
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc8899700 (LWP 7564)]
__memcpy_sse2_unaligned () at ../sysdeps/x86_64/multiarch/memcpy-sse2-unaligned.S:33
33	../sysdeps/x86_64/multiarch/memcpy-sse2-unaligned.S: Aucun fichier ou dossier de ce type.

3djc, 2016-11-18:
ok, steps to reproduce:
1) create or load a model with a replace mixer
2) view the mixer either in blue or default
3) go back, load the other theme
4) go to view the mixer page
5) segfault

3djc, 2016-11-18:
static const BitmapBuffer * mpx_mode[] = {
    mixerSetupAddBitmap,
    mixerSetupMultiBitmap,
    mixerSetupReplaceBitmap
  };

It is because of this static

projectkk2glider, 2016-11-18:
Perhaps while you at it, add a check that prevents writing/drawing outside the LCD buffer. This is already done on Taranis, but needs to be done for Horus and small pixel screen (ie sky9x).

Githubbachelor, 2016-11-18:
You have that problem if you switch through the Menu sites. If you switch to quick the Horus hang also up.

3djc, 2016-11-18:
This issue will be fixed in the next nightly. In the interim, avoid changing theme ;)

commit 6c006c3e (2016-11-18): Horus/fix #4036 (#4038)

* Remove 'static' from bitmapbuffer that where causing issues when changing theme. I have checked that this does not create extra sdcard access

* Cosmetics

* Move calibration bitmaps to theme

* Overlaping PR

* There no theme changing during shutdown

* Cosmetics
closed 2016-11-18 by 3djc