opentx/opentx #4039: FrSky Taranis Plus - writing on SD card.

Georgi65, 2016-11-18:
Hi I have problem and I have run out of ideas.
I have OpenTX 2.1.9 (2016-09-15) I updayted also bootloader. to the same ver.
I start Taranis in Bootloader and connect cable USB. 
I tried 4 cables on 3 Computers. 2 laptops - usb 3.0 and one PC usb 2.0 All have windows 10 
Windows see Taranis as 2 things and one of this is Memory storage. 
I Can copy from the storage. But I can't Copy to CD Card. 
When I want to do this my taranis it freezing. 
I updayted firmware by Companion 2.1.9 with 0 problems. The only one problem is when I want copy something to card. Can someone help me ? What should I check, change ? 
Is this Taranis problem ? or my cables/PC drivers ? :(

3djc, 2016-11-18:
If not cable, I would test the sdcard, a likely candidate for failure too

Georgi65, 2016-11-18:
Ok will try that but when I take out card and put to card reader then I can copy files with no problem.

3djc, 2016-11-18:
We need more data then to confirm. You say you cannot copy to, but can you copy from ?

Georgi65, 2016-11-18:
Yes when I connect from bootloader taranis I can copy FROM the SD but not to SD. 
When I connect the same SD to card reader then I can copy to SD.

Georgi65, 2016-11-18:
THANKS ! I changed SD card to 4GB and FAT32 Format. And it works !!!!

closed 2016-11-18 by Georgi65