opentx/opentx #4041: Screen Problem at Choosing Model

Githubbachelor, 2016-11-19:
I copy a ready made model Polizia in a folder Heli. Than I changed some data in the mixer, make a new name DARIO 600 PRO D8 prepare all. After you will load again a model from the folder Heli (in my example) you see a wrong sreen from that model DARIO 600 Pro D8. And you cannot read the name in the screen. See pics.

[image]
[image]
[image]

Githubbachelor, 2016-11-19:
You can move the Model but the hacked screen from picture and name is still there. If you make a next new model it is clear.

3djc, 2016-11-19:
I cannot reproduce this issue, (please join sd card backup), but I could see a minor one when model name is too long:
[image]
[image]

Name is cut

bsongis, 2016-11-19:
I also think that the name length is the reason of the issue. We should add some protection to avoid writing  outside the BitmapBuffer (the one created in modelslist.h:52)

3djc, 2016-11-19:
@bsongis : I have a proposed patch (https://github.com/opentx/opentx/commit/b0df773c03090ba0fdbc5b13ac1eb07e27d23ee0), let me know what you think

bsongis, 2016-11-19:
You also may write with a smaller font?

3djc, 2016-11-19:
I sure can, here is how it looks:
[image]

3djc, 2016-11-19:
But even smlsize may noit always be enough:
[image]

3djc, 2016-11-19:
This is smalsize + char number cut

[image]

3djc, 2016-11-19:
Even with smallsize, we can fit only 13 out of the 15 chars allowed for model name when using the larger chars

3djc, 2016-11-19:
I'm rearranging the whole screen, stay put :)

3djc, 2016-11-19:
[image]

Githubbachelor, 2016-11-19:
Hi djc and Bertrand
do you have a actuall patch from now whithout deleting anything in my ready models , which I can install? And how I must do? Where?

3djc, 2016-11-19:
If my patch is accepted, it will be available in the next nightly

kilrah, 2016-11-19:
The only thing you'll have to do is update when the next RC is out. Leave those models alone until then.

Githubbachelor, 2016-11-19:
ok

Githubbachelor, 2016-11-19:
By the way, are the internal multi protocols in OpenTX working with the multi modul from Banggood Version2 WITHOUT PPM ? Someone told only to make a bridge in the multimodul and switch to zero Position, and it works. Did someone test that? 1000 people write anything without a 100% result. And a very important protocol is the Nine Eagle protocol which is written but nowhere to find?

kilrah, 2016-11-19:
Yes they work. Selections may not include the very latest things that were added to the multi module, but you can always use CUSTOM to reach those until they get added.

Githubbachelor, 2016-11-19:
Yes I want to find that Chat but where? Do you have a link for that. I am still new in Github and only find that here

Githubbachelor, 2016-11-19:
I would like to have a contact to you outside from here, over email if possible from Bertrand, Andre 3djc, kilrah. How can I write to you ?

bsongis, 2016-11-19:
Here:
https://opentx.rocket.chat/channel/OpenTX_General

3djc, 2016-11-20:
This is fixed in next RC.

commit a35d6b6a (2016-11-19): 3djc/horus fix #4041 (#4042)

* Reduce model name size until they fit into the buffer

* SMLSIZE is better (but cut is still needed)

* Rework screen layout to handle up to 15 chars

* Allow for the two scroolbars (models and cats)
closed 2016-11-20 by 3djc