opentx/opentx #4043: Taranis 19-11-2016   Mixer screen cut off on left

fraternl, 2016-11-19:
The mixer screen is cut off at the left
[image]

3djc, 2016-11-19:
I reopened because it looks there are more issues on this screen than just this one

fraternl, 2016-11-19:
[19-11-2016-all-models.eepe.txt](https://github.com/opentx/opentx/files/601399/19-11-2016-all-models.eepe.txt)

fraternl, 2016-11-20:
On today's compile (20-11) the mixer screen is still garbled.
The global variable is now gone, but much stuff is being partially overwritten by other...
Taking the mixer screen of one of my DLG's as an example should cover most of those.

@bsongis made a change yesterday afternoon, but it wasn't attributed to this ticket.

[image]

bsongis, 2016-11-20:
The changers were pushed on a dedicated branch to allow somebody else than me to test it before it's ported to 'next'. Now this has been done and everything should be back

bsongis, 2016-11-20:
I will be glad to have another report and be sure everything is nice ;)

closed 2016-11-20 by bsongis

fraternl, 2016-11-20:
yes....   I can't see anything wrong now.
Will inspect some more...

PR #4045 (2016-11-19): Revise column for mixer
This fixes #4043
commit e14d04b5 (2016-11-19): Revise column for mixer (#4045)

This fixes #4043