opentx/opentx #4044: Taking screenshot does not work using ENT - EXIT

fraternl, 2016-11-19:
Taking screenshot does not work using ENT - EXIT
I don't know if it ever worked, but it seems it did as I could find several references to it.
http://rc-soar.blogspot.nl/2015/07/opentx-21-preview.html

I needed it to file the previous github issue.
Now I used a global function.

bsongis, 2016-11-19:
Would we remove the special function "Screenshot" then? it wouldn't be needed anymore

kilrah, 2016-11-19:
No, it is still useful in flight! I tend to screenshot my telemetry page at the same time I trigger voice announcements with SH, gives status snapshots to have references

bsongis, 2016-11-23:
Ok. The [ENT] + [EXIT] remains a good idea, do you agree?

kilrah, 2016-11-23:
Yes!

projectkk2glider, 2016-11-23:
On Horus pgup+ent would work better I think.

closed 2020-08-11 by stale[bot]