opentx/opentx #4054: Very Important - Please set SLIDER RS & LS FROM HORUS INTO REVERSE

Githubbachelor, 2016-11-19:
If you take a Slider RS or LS and you slide it to your body than you have the maximum. That is exactly a reverse to Taranis Plus, where slider RS & LS have a minimum. 

Please change in HORUS OPENTX 2.2.0 359/360 coming, and set the Slider RS and LS into reverse.

3djc, 2016-11-20:
There is no right or wrong given the position of the slider. It has been decided a long ago to work that way. If you prefer it in the other direction, just put a negative weight.

closed 2016-11-20 by 3djc

Githubbachelor, 2016-11-20:
Hallo Djc
I want to come back again to this point. It is important for all user who change from Taranis to Horus.

In Taranis the LS is in position to your body and the position in Screen monitor is -100%. Away from you it has position right in the sreen Monitor and has+100%.

in the Horus the LS is to your body (IN DOWNPOSITION) and the position in Screen Monitor is NOT down -100%! It is the Top +100%. Same with RS.

The problem is that if you build your models, and Bind them that for example Flaps BRAKE - if you do not think to do the reverse for that channel. That is a VERY BIG DIFFERENCE to all Horus user.

I hope you understand what I mean, because my englich is not so good.

If change that in the Firmware OPEN TX ONLY for Horus. So to do is a REVERSE in the Program for LS and RS and a Change in the Sreen for the position from UP to DOWN.

Greetings Michael

kilrah, 2016-11-20:
Horus is a different radio. It has different controls. Someone transferring a model from a Taranis will have to rethink all of the auxiliary inputs anyway, maybe he'll rather like what he had on the side sliders in one of the front ones. Maybe he'll leave it at the back, but the different ergonomics mean he'll want it the other way.

There is no reason to expect it to be the same. There however now IS a very good reason not to change it, since everybody who already has a Horus model set up would be at risk.