opentx/opentx #4056: REOPEN! IMPORTANT SLIDER LS & RS FROM HORUS - Sorry but I think it is nessessary

Githubbachelor, 2016-11-20:
In the Horus """FRSKY OS"""" is that ok!!! And in X9E 2.1.9 OK.

Only in HORUS OpenTX Wrong 2.2.0 NB359 NOT OK!

Can you have again a look for this? Thanks Beforehand
Very Important - Please set SLIDER RS & LS FROM HORUS INTO REVERSE in 2.2.0 /360 #4054

Hallo Djc
I want to come back again to this point. It is important for all user who change from Taranis to Horus.

In Taranis the LS is in position to your body and the position in Screen monitor is -100%. Away from you it has position right in the sreen Monitor and has+100%.

in the Horus the LS is to your body (IN DOWNPOSITION) and the position in Screen Monitor is NOT down -100%! It is the Top +100%. Same with RS.

The problem is that if you build your models, and Bind them that for example Flaps BRAKE - if you do not think to do the reverse for that channel. That is a VERY BIG DIFFERENCE to all Horus user.

I hope you understand what I mean, because my englich is not so good.

If change that in the Firmware OPEN TX ONLY for Horus. So to do is a REVERSE in the Program for LS and RS and a Change in the Sreen for the position from UP to DOWN.

And if you do a reverse, the main screen does not change.

Greetings Michael
I mean if you do the same in Horus OpenTX as in the Taranis OPENTX you have reverse in the LS/RS Channel and a wrong Mainscreen showing the Position, Down is Top and Top is Down.
The problem was yesterday if I take all my date from a Model in Taranis to a new model in Horus. I prepared all. And if I want to bind that model my flaps go in wrong way and broke. Thats why I tell you this.

Githubbachelor 14:39 
In the Horus """FRSKY OS"""" is that ok!!! Only in OpenTX wrong 2.2.0 NB359

3djc, 2016-11-20:
There is no point in re-opening another, similar, one.

You express a logical point of view, however, another, logical as well point of view was expressed when the decision was taken. As I explained, there is no right or wrong, just different spin on it.

That decision was made before I started helping the team There are now thousands and thousands of Horus from beta  to production units, and 10 or more times that as tested, validated, flying, rolling, boating,.. models.

That decision simply cannot be changed at that point in time, no matter what x, y or z, is thinking about it

closed 2016-11-20 by 3djc

Githubbachelor, 2016-11-20:
OK than I have to change the cables

kilrah, 2016-11-20:
It cannot be "ok" in FrSky OS - it'sn an entirely different system, in it you have configured your model from scratch, and obviously set it right there. Can't be compared.

You simply have to adapt your model setup to the new radio.

Githubbachelor, 2016-11-20:
OK I have changed the cables in the connector from the potis LS and RS. Now all is ok.