opentx/opentx #4057: splash screen resolution

hketterer, 2016-11-20:
i have a custom splash screen that was readable in 2.1.8 but unreadable in 2.2 seems like the resolution was downgraded.

kilrah, 2016-11-20:
Maybe specify what radio type?

hketterer, 2016-11-20:
sorry, taranis plus

jtaylor2, 2016-11-20:
I noticed the same Taranis splash on 2.2 didn't look as good as it did on 2.1 but hadn't had time to look into it.  It seems that Companion 2.2 does something not nice to the profile picture and also if you elect to use a different picture when you download the firmware.  Here are captures of 2.2 and 2.1 splash editing screens and also 2.2 and 2.1 download windows.  These are all the same picture.
[image]
[image]

bsongis, 2016-11-21:
obviously the splash is converted to 1bit

jtaylor2, 2016-11-22:
The problem is that the taranis to x9d name change missed a couple of spots and the x9d splash images are being formatted for a 9x.  I will generate a pull request to fix it.

jtaylor2, 2016-11-22:
As @kilrah  pointed out fix does not cover X9E.  I will do another pull request to correct.

closed 2016-11-22 by bsongis

bsongis, 2016-11-22:
@jtaylor2 would you have a look to this PR:
https://github.com/opentx/opentx/pull/4067