opentx/opentx #4060: Bug? HORUS OPEN TX 2.2.0 359 PROBLEM Missing Channel 10 in Screen

Githubbachelor, 2016-11-21:
In the Mixer Screen channel 10 is missing. See picture. That is a bug I think. Channel 1-14 is used in my configuration. 10 is not shown.

[image]

3djc, 2016-11-21:
You have to choose the first chan in the widget settings

Githubbachelor, 2016-11-21:
I have set the first Channel

3djc, 2016-11-21:
I may fix it, but this widget is not designed to be used on this screen

3djc, 2016-11-21:
[image]

3djc, 2016-11-21:
I may also remind that you have a much more evolved chan monitor with long push on the wheel and monitor

Githubbachelor, 2016-11-21:
Can I take your screen? I do not understand?

3djc, 2016-11-21:
you have put the chan mon on a layout not adapted for that, you need for example to take the layout on two collums like I have done.

Githubbachelor, 2016-11-21:
OK let me test

Githubbachelor, 2016-11-21:
Ok I try this in 2 Windows I can set the second Window from Channel 13 upwards.

But If you take 1 big screen, it has to check automatic from channel 1 to 20

closed 2016-11-21 by bsongis

3djc, 2016-11-21:
This will be fixed in the RC

PR #4061 (2016-11-21): Fix two column handling
This fixes #4060