opentx/opentx #4062: IDEA! HORUS- IN THE MAIN MENU ROUTE VOLUME TO A POTI OR SLIDER-Maybe in NB360

Githubbachelor, 2016-11-21:
PLEASE PLEASE PLEASE Guys

Is it possible to build a point in the Main Menu where you can ROUTE THE MAIN VOLUME to a prefered:

Switch, Slider, Poti. That is a very big help to save the nerves.

Also to build in the next Genaration HORUS B directly a poti into the SpeakerCable.

kilrah, 2016-11-21:
The Volume special function has been there since day 1 of the Taranis.

Please validate your requests before posting.

closed 2016-11-21 by kilrah

Githubbachelor, 2016-11-21:
I know that, but you did not understand me. I do not mean to program that in every Model. I mean a one time done for future. I did not find that in the main menu!

kilrah, 2016-11-21:
Global functions.

bsongis, 2016-11-21:
Also please stop using the private rooms of our chat to explain your idea. As stated above IT'S ALREADY IMPLEMENTED.