opentx/opentx #4063: Little Issue with lua mix scripts in Companion - Taranis 2.1.9

pau33, 2016-11-21:
There is no special page to introduce lua mix scripts in Companion. But you can adress them in the Companion Simulator. When you do so and also when you change the name field there, this will not be saved with your model (no * that model parameters have changed). The only way to save the scrips and name tags in Companion is to set the scripts and the name field in the real radio and then import the settings to Companion, but not vice versa. Keeping version control in the lua-scripts is not really easy this way. Additionally controling the number field from lua would be good, I adressed this in #3687.
Not a to high prior point, but should be considered some time.

kilrah, 2016-11-21:
You can use the standalone simulator to have settings changed.

Mix scripts can't be handled in companion becasue the behavior depend on the script contents - i.e. companion would need the capability to "execute" the script to know what settings to display. 

Duplicate of #1488

closed 2016-11-21 by kilrah

pau33, 2016-11-21:
This was a little bit too fast for me. In my case Companion can handle the mix scrips in the internal simulator (from the telemetry field?) and the external simulator (from the model page?). Both simulations work, but when I want do modifications of the lua name or the name tag (version control) in both simulations, they will be not saved with the model. They will be saved, if doing this in real Taranis and importing the model settings form there. So I do not really understand the difference. Maybe you can comment this in detail.
Thank you in anticipation.

kilrah, 2016-11-21:
Edits done in companion simulator are NEVER carried back, it's only for viewing.

You can however work on a copy of your eeprom with the standalone simulator, not in companion, separate program called "Firmware Simulator 2.1" in your start menu.

lshems, 2016-11-21:
Chiming in, as it had me confused as well.

With the seperate simulator, you actually access an EPEE as would the real transmitter do. So changes are saved to the EPEE, and you can use the scripts stuff as well, set it up, and save the model to the EPEE.

My normal mode of working is to make changes using the simulator, and save the changes from the simulator by making a backup in the model setup page, as if you where on the actual radio.

Then you can simply open or import the saved model in companion.

Be aware, the simulator uses the settings of companion for finding your SDcard path!

P.S. since my scripts have become quite powerfull to set up new models, I even use my scripts in the simulator to create and setup new models, as if I where on the radio. Really funny, that it is quicker this way, than in Companion itself. But it works just because I accept using  a standard set of features every time.

pau33, 2016-11-22:
Thank You all, now I got it. The Simulator works with a backup of the model.
With that in mind you can close #3687. It won't work the way I thought.