opentx/opentx #4069: At loading another model happen NOTFALLMODUS / EMERGENCYMODUS

Githubbachelor, 2016-11-24:
Hallo since an h iam preparing my new models. 5 minutes ago I want to load another model and the system was going into NOTFALLMODUS with a short loud crack. must be a Bug. Please tell me what you need to find out. See picture
[image]

bsongis, 2016-11-24:
Is it reproductible? I mean if you go to the other model and reload this one, do you reproduce it?

Githubbachelor, 2016-11-24:
No sorry not reproduc. I must switch off. Than Switch on and all is working again. Sporadically

Githubbachelor, 2016-11-24:
Now I go on installing my models. Maybe come anothertime??? Hopefully not. I think it has to do with storing. At the moment if I hold the button to load came the crash. What happen at this moment in the firmware? Maybe to find there.

Githubbachelor, 2016-11-24:
button to load a MODEL

projectkk2glider, 2017-01-14:
Closing for now, reopen if you can reproduce this issue.

closed 2017-01-14 by projectkk2glider