opentx/opentx #4070: Important IDEA for next UPDATE Please make Telemetry Screens 1-max5 Moveable

Githubbachelor, 2016-11-25:
ebver time when I prepare my Screens for Telemetry and when they are ready, I would like to change the screens between the positions 1-5 max. For example move Screen to position 4 and 4 to 5. If yoou wnat to do that now, you have to delete complete screens and you have to make them new.

bsongis, 2016-11-25:
The idea was already given by @mhotar and he even made the icons for that.

Githubbachelor, 2016-11-26:
What do you think, it is possible for 360NB?

lshems, 2016-12-14:
If the script could be set from LUA, there would be no need for this. A verty simple LUA script could then do what you want.

see https://github.com/opentx/opentx/issues/3620

lshems, 2016-12-14:
Sorry, o,n second thought thta doesn't answer your question. that would only work for the telemetry scripts itself. we would need full acess to all telemetry page variables to be shown to do this in LUA.

Actually, I am thinking of creating a LUA script to mimic the standard telemetry output screens, but allow for much more flexibility. For instance the size, icons used, location on the screen, etc. could all be freely assigned.

projectkk2glider, 2016-12-14:
@lshems there are no telemetry scripts on Horus - it uses different kind of scripts - widgets.

Githubbachelor, 2016-12-14:
I only want tomake READY MADE Screens to move from position 5 to 4! Or from pos. 2 to 3 and so on. Nothing more. Same as moving Models to other positions in the folder.

lshems, 2016-12-15:
Ok, I do not know anything about Horus. So if it's about that, i've not said anything.

As far as ready made screens, it is rather cumbersome to have to specify the entire setup, for each screen seperately, and not being able to move them. (but writing in capitals is not neccesary to make your point)

I could even imagine some people would prefer this to be a radio wide setup, as they use the same telemetry unit(s) on several or all of their models. I am no fan of that, nor of general functions, which aim to achieve something similar. 

When adding to much functions that are not obvious, or rarely used, things become overly complicated.

Just an opinion, so you are free to agree or disagree ;)

kilrah, 2016-12-15:
Nothing of this realtes to the actual question/issue. Completely irrelevant.

On Horus you can configure screens, and currently you can't change order of existing screens, if you want to do so you need to reconfigure them completely manually.

lshems, 2016-12-15:
"Nothing of this relates to the actual question/issue. Completely irrelevant."

thanks for the opinion.

Githubbachelor, 2016-12-15:
Ok I see. So there is no way to move ready made Telemetry screens from their positions. Ok so we have to do manually all new if we think, the first produced sreen to see on position 3 for example. Thanks for your answers.

closed 2020-04-06 by stale[bot]