opentx/opentx #4071: 2.2N359 Assign Brightness to any available pot or switch

Joetrin, 2016-11-25:
Not really an issue, but I'd like the ability to adjust screen color using any assignable knob or switch.  I guess this can be considered an enhancement.

Thank You

SimonJux, 2016-11-25:
This would be great to adjust screen brightness and volume on the knob

schwabe, 2016-11-25:
See also https://github.com/opentx/opentx/pull/3988
but that pull request is not yet fully ready.

Viruswelt, 2017-04-12:
Maybe it would be possible to adjust screen brightness for each model separately?
When I fly simultor I need a different brightness than in the hall (gym) or on the airfield
especially for the Horus

Joetrin, 2017-04-12:
Adding to @Viruswelt , if it can be applied in Global Function and Special Function, then we could have both options.

Thank You

poge-st, 2018-03-22:
Could be solved like the Volume feature with an Special Function/Global Function "ON" - "Brightness" - Input (Switch, Knob, .... )

der-ray, 2020-02-23:
Having a global function for brightness adjustment via poti would be great for radios with a big color screen like the horus. FPV users for example dont need the screen brightness to be up all the time but only in bright sunlight when looking at it.

ikcalB, 2020-03-28:
would also love to see that implemented!
same as Volume, like @poge-st mentioned?

PR #7680 (2020-07-31): Brightness adjust from backlight function
This PR closes #4071 and #6163

Adds a source to backlight function to control the brightness:

- Source: if assigned, controls the brightness. If not assigned brightness in radio configuration is used
- Switch: if assigned to ON, auto off as defined in radio config. Else light on when switch on (as current behaviour)

If assigned to switch ON the function is always active. Due to this it is better to have enable option but this would require an eeprom change. Enable option can be added in 2.4
closed 2020-07-31 by bsongis