opentx/opentx #4072: Loosing/Changing Model-windows in the folders. Pics going Picasso?

Githubbachelor, 2016-11-26:
Yesterday night about 1h30 I programed my Model 64. When I was ready, I wnat to load another Model to control if all is working, cause sometimes the sytem was hanging. But this time my Model Pictures with names become multicolorpixels and unreadable. When I go back in my ready made model with exit the model was ok. Than I try to load another time and in my folder more and more mocelpics become not readable mixed color windows . If you scroll over the models they become more and more unreadable windows. I was afraid that I loose all my ready made models. So I hold the Switch Off button to close. After new starting th TX all was OK, hopefully. Also I found out that the most hanging problems happen when the battery from TX goes to end of using, but that is not sure. I think so??!! But again not reproduceable, that is the most biggest problem.

kilrah, 2016-11-26:
Looks like your SD card is faulty and that is causing all your issues.

Githubbachelor, 2016-11-26:
I have 3 SD card, doesnt matter which card, on every was a problem. They are the quickest cards you can get. I thin it has to do with Voltage at the end of using the Horus.

Githubbachelor, 2016-11-27:
Last night again same procedure but not reproduceable. look pics. This happen if you have made a new model and you are ready with this. Than you want to make the next model and if the window come with the folder for models one picture has that. If you start to scroll in the folder over the ready made models came more. If you choose the next folder all models hace that pics abo. Look pictures. After you have to switch off the TX and switch on again and is ok.

[image]
[image]
[image]
[image]

kellybyrd, 2016-12-12:
Could this be related to #4099 ?

Githubbachelor, 2016-12-12:
No these are two different things

projectkk2glider, 2017-01-14:
I will tag this as duplicate of #4099.

closed 2017-01-14 by projectkk2glider