opentx/opentx #4074: sound files starting with underscore are not shown on Taranis, but still on Horus

fraternl, 2016-11-26:
To let a lua-script tell the difference between normal sound files and music I have all my music placed in /sounds/en/ starting with an underscore.
In current 2.2 builds these files can't be selected as a sound file even though they are in 6.3 format.
On the Horus these files can still be selected.

Githubbachelor, 2016-11-27:
I had the same problems. I think you must rename your files and give not more than 6Letters. I take a 0 at the Beginning and they work.

fraternl, 2016-11-27:
Almost...
The way things are sorted is changed.
It wasn't able to find the song that was already selected and it jumped to beginning.
This used to be underscores and is now numerics.

closed 2016-11-27 by fraternl