opentx/opentx #4080: Possibility of Preinstalling for example RSSI, Curr, A2 aso. in Special Function

Githubbachelor, 2016-11-29:
We need the Possibility of Preinstalling for example RSSI, Curr, A2 aso. in Special Function, without binding a Receiver. So we can prepare our models before using. So that we can set the functions of telemetry data in SF. I hope you understand what I mean.

schwabe, 2016-11-29:
Lua scritps already have this functionality. If you want to test/simulate your model with telemetry the simulator on the PC can already do that.

Manually adding telemetry is already error prone. What is the problem of just binding a receiver?

Githubbachelor, 2016-11-29:
I have 80 models. And I want not go to take every Model from my store to bind. I want to prepare every model with data from Taranis+ and bring these data manually to HORUS 2.2.359. And for to do this I need the option to prepare in SF "RSSI say Data RSSI 15s". And I cannot choose RSSI or A2 Curr, if I do not bind a receiver and search for Sensors.

3djc, 2016-11-29:
You should not be able to, as the radio as no way to know before binding what the actual model (receiver + probes) will be capable of

Githubbachelor, 2016-11-29:
Ok I understand but I know which receiver will be installed and which function/sensors the Horus will recognize later. I know that the system can not tell data actually, only if I bind. But I only want to prepare all. But if not possible than it is so.

3djc, 2016-11-29:
Then you might want to copy a model with  corresponding receiver and start from there

schwabe, 2016-11-29:
If you have 80 models lying around, there is a good chance you have a spare of all sensors lying around. Just connect the rx+the sensor on your bench to do the discovery.

No I am not Helmut Renz 😕

Githubbachelor, 2016-11-29:
Thanks Good Idea

closed 2016-11-29 by 3djc

lshems, 2016-11-29:
Where can I find info on how to set sensors from LUA? Which opentx? Thanks.

schwabe, 2016-11-29:
@lshems that is just lazy. telemetry lua in the issues or pull request would have given enough hints to discover the function in lua or the pull requst (e.g. #3977)

lshems, 2016-11-30:
Lazy I am, but not on this part. Ive been searching, and didn't find it. I have real difficulties understanding this github stuff when it comes to the code part.

Sorry if I ask questions, didn't know that classifies me as lazy.

RealTadango, 2016-11-30:
https://github.com/opentx/opentx/tree/next/radio/src/lua

3djc, 2016-11-30:
The link to the change request gives all the details ....

schwabe, 2016-11-30:
@RealTadango setTelemetrySensor will also add sensors.