opentx/opentx #4084: Sticky glitch on v2.1.9

RightRudder, 2016-12-01:
Create a L1 logical switch as sticky. 
 In V1 SH (down)
 In V2 SG (mid)
 In 'AND Switch'  SG (mid)
Simulate.
 Move SG(mid) pull SH (down) see L1 goes on and stays on when SH released.
 Move SG (up) see L1 goes off.
 Move SG (mid) see L1 goes momentarily on. (glitch)

projectkk2glider, 2016-12-01:
The last change that affects this was here https://github.com/opentx/opentx/issues/1646#issuecomment-53879076, commit https://github.com/opentx/opentx/commit/6195f5ecc1667f373126ce9e7e611a24906572d4

projectkk2glider, 2016-12-01:
The best solution to this is to avoid such setup. I think we will not change anything.

Here are the details why this is happening:
 * state change of V1 or V2 for STICKY switch is checked in a timer tick every 100ms.
 * actual LS final state is processed/calculated when needed.

So in the second part when you `Move SG (up) see L1 goes off.` the internal LS state stays TRUE (because V2 has not been triggered), but since the AND switch is now FALSE the end result is FALSE.

Then you `Move SG (mid) see L1 goes momentarily on.` The LS state will become FALSE after 0 - 100ms, (when the timer tick gets fired). If you try to read the state of the switch before the timer tick arrives (which is normal, as switches are read ever couple of ms), you will get TRUE (because both AND and internal state are TRUE).

I am sure you can find a better solution, using the same condition for OFF triger and AND condition looks  strange. What are you trying to do anyways?

lshems, 2016-12-01:
There is an extended debate here, covering more or les sthe same topic:

https://github.com/opentx/opentx/issues/3261

Guido

RightRudder, 2016-12-01:
Thanks for your response,

I want to use the momentary switch as a way to prevent inadvertently 
entering a PID gain setting mode for a stabilizer connected to the RX.  
To enter gain setting sw A has to be down then press momentary SH which 
latches the gain setting mode on.  When SWA is removed from POS3 it 
exits gain setting.  It should no be re-entered without the momentary 
confirmation.   Is there a better way to do this?

RightRudder, 2016-12-02:
So how can one make a glitchless latching switch?

RightRudder, 2016-12-02:
I tried adding a 0.1sec delay.  The glitch is gone.

projectkk2glider, 2016-12-03:
Your delay trick is a nice solution. Another solution is to use two logical switches:
```
L1    AND    SHdown   SGmid
L2    STICKY   L1    !SGmid
```

closed 2016-12-03 by projectkk2glider

RightRudder, 2016-12-03:
Thanks for that.  There are many roads to the truth especially with such an awesome system as open taranis.