opentx/opentx #4086: X9D Crashing upon plug in while turned on for simulator

Fickert07, 2016-12-02:
I keep getting my Taranis to crash while plugging into my pc to fly in DRL. I updated from 2.1.9 to 2.2.0N359 and it worked before. I checked all the drivers as far to my knowledge but with still no luck and almost always crashes when plugged into the pc. This is a new pc but have tried installing and uninstalling various drivers without any luck.

Any Idea?

schwabe, 2016-12-02:
I cannot reproduce the behaviour with either Mac or Windows whith m Taranis+ and 2.2.0N359. Could add more details like OS, build options etc? Did you try it on antoher PC? Did you work for you in past. If yes, which version?

Fickert07, 2016-12-02:
My previous version was 2.1.9. This was the first attempt on this new pc but I flashed the latest 2.20N359 on this with this new pc without issues. I use the Zadig (I think?) program to install the correct driver, but  I even deleted that in hopes of it working but nothing thus far. 

DRL Recommends a driver to download from their site to use but when I try to get my taranis to install it and use it through the device manager it says that it fails. And I have noticed if the Taranis isn't crashed trying to install drivers is when it does crash.

bsongis, 2016-12-02:
What are the options you choosed for your firmware?

Zadig is only used for flashing your radio, not for massstorage / serial / hid USB classes

projectkk2glider, 2016-12-03:
Using `cmake -DPCB=X9D -DPCBREV=OFF -DCMAKE_BUILD_TYPE=Debug  -DDEBUG=YES -DCLI=ON -DDEBUG_TIMERS=YES -DDEBUG_INTERRUPTS=YES ../../opentx` for testing.

So far (on latest commit + couple of minor fixes to serial2 driver) + Linux on host:
 * Joystick works - connected/disconnected multiple times
 * Bootloder USB connection  works - copied from the SD card more than 50MB of files, read/written models, wrote 30MB file to the SD card. Also  connected/disconnected multiple times

projectkk2glider, 2016-12-03:
Using `cmake -DPCB=X9D -DPCBREV=OFF -DCMAKE_BUILD_TYPE=Debug  -DDEBUG=YES -DCLI=ON -DDEBUG_TIMERS=YES -DDEBUG_INTERRUPTS=YES -DUSB=MASSSTORAGE ../../opentx`:
 * connected USB
 * read/written about 40MB of files - no problems. Also connected/disconnected multiple times.
 * read/written models - no problems

projectkk2glider, 2016-12-03:
Using ` cmake -DPCB=X9D -DPCBREV=OFF -DCMAKE_BUILD_TYPE=Debug  -DDEBUG=YES -DCLI=ON -DDEBUG_TIMERS=YES -DDEBUG_INTERRUPTS=YES -DUSB=SERIAL ../../opentx
`

Confirmed crash, steps to reproduce:
 * turn on radio
 * connect USB cable
 * open virtual serial port
 * the debug prints work and radio is stable
 * close serial port (without disconnecting USB cable)
 * force debug print (like pressing MENU button) - radio resets and ends up in STM Bootloder because USB cable is still connected.

projectkk2glider, 2016-12-11:
This should be fixed by the #4097

PR #4097 (2016-12-06): Projectkk2glider/issue 4086 usb and serial
Closes #4086

I tested this on the Taranis in all three USB modes. Tests on other radios (X9E and Horus) have to be done before the merge.

The change is big, but the only way to solve the USB=SERIAL WDT resets was to use the parts of the newer ST USB library. Then I decided to port the whole library while preserving our changes and fixes.
closed 2016-12-11 by projectkk2glider