opentx/opentx #4095: DIY Spektrum Module doesent work anymore since OpenTX 2.2 RC6

rcbebo82, 2016-12-04:
I built a DIY Spektrum module with a module from a DX5e. Im using a horus with OpenTX 2.2. Binding was fine with RC5 after RC6 update it does not work anymore. It also tried RC8. Binding is not posibble in any Spektrum mode. I tried with a OrangeRX R410x and a Storm S603.
On my Taranis X9D+ with OpenTX 2.1 it works like a charm.

RealTadango, 2016-12-06:
I use the exact same DIY module and it works fine with RC8 in my Horus. When switching back to RC5 does it work again?

rcbebo82, 2016-12-07:
I will downgrade this evening and test it again.

schwabe, 2016-12-07:
If it does I can take a logic analyzer and check if there is any difference.

rcbebo82, 2016-12-08:
Reflashed RC5 yesterday. Does also not work 8-(, anyone any idea?

schwabe, 2016-12-08:
That sounds like either your taranis output is broken or your DIY module. If you have an osciloskope/logical analyzer you can check the signal. Otherwise try with a DMM if the voltage are what you are expecting on the DIY module.

RealTadango, 2016-12-08:
Since it is working fine for me i suspect your module bay is broken. Could be the power or the signal that is not working. Do you have another module you can test?

rcbebo82, 2016-12-08:
I will measure the voltage from the Horus Pins. The module works well in my X9D so i think the module is fine. Another module is not available but i can build one. I have a DMM but no osciloskope. I will compare the voltages from X9D and Horus. Mayby the module bay is really broken this would be very sad.
 
Be back this evening with new informations.

RealTadango, 2016-12-08:
And you have tried another clean model? Just to rule out a weird bug......

rcbebo82, 2016-12-08:
Well no didnt try that, will check that also.

I have flown a small Nurflügel in my holidys with a R410x and my horus (RC5). After i came back i flashed RC6, last weekend i took the Nurflügel to my flying place and it was not bound anymore. I then tried to rebind it but it doesent work.

RealTadango, 2016-12-08:
You could test RF output of the module by listening to the speakers. There is a little noise when the module is transmitting. The same as with the internal module.

rcbebo82, 2016-12-09:
Horus: Output voltage from jr bay 10,4V. Voltage to DX5e Module 3,4V noise is heareable.
X9D: Output voltage from jr bay 7,4V. Voltage to DX5e Module 3,39V noise heareable.

It works again. I think my antenna cable has a problem. Will check that now.

closed 2016-12-24 by kilrah