opentx/opentx #4098: Horus OpenTX 2.2.360 RC8 Problem with Model Name?

Githubbachelor, 2016-12-05:
Hallo
A few minites ago I make a new model and it got the name "model.5" automatically. After I opened an other ready made model. And than I want delete model.5 and I go to delete. The screen ask me to delete model.125? Than I delete it and "model.5" was gone. What is about the difference Model.5 =model125?

Please look pictures.

[image]
[image]
[image]

3djc, 2016-12-05:
One is the name of the model (Model 5), the other one is the name of the file holding the model data. I leave this open until we decide if we should display model name on your last screen

bsongis, 2016-12-05:
We should perhaps display the model name + the model filename, no?

3djc, 2016-12-05:
Yeap, will do that today

3djc, 2016-12-15:
Fixed in next nightly

closed 2016-12-15 by 3djc

Githubbachelor, 2016-12-16:
Thanks a lot!

PR #4100 (2016-12-15): Deletion confirm changed from filename to modelname(filename)
Deletion confirm changed from filename to modelname

Fixes also Taranis that was not displaying model name

This fixes #4098