opentx/opentx #4099: Horus OpenTX 2.2.360 RC8 Problem with Screen again

Githubbachelor, 2016-12-05:
Horus OpenTX 2.2.360 RC8 Problem with screen again after laoding some models. Maybe same problem as known? But I dont know. Because new Version 2.2.
[image]
360

bsongis, 2016-12-05:
Is it 100% reproductible? Would you attach the model pictures?

3djc, 2016-12-05:
There is an issue with cat lenght at least, when I testes I could not put more than 9, en know it's 15, which goes way beyong thr colunm

3djc, 2016-12-05:
[image]

3djc, 2016-12-05:
@Githubbachelor : please include a full copy of your sd content

Githubbachelor, 2016-12-05:
OK I make a zip for uploading now

Githubbachelor, 2016-12-05:
Cannot upload is too big the zip

Githubbachelor, 2016-12-05:
where can I upload

kilrah, 2016-12-05:
It's up here: http://www.filedropper.com/sdkarteanlagehorusaktuelleversion360_1

3djc, 2016-12-05:
The zp provided contains 92 models, the screen above show 93 ...

Githubbachelor, 2016-12-05:
Yes thats right I killed the wrong new model.5 which was model125.bin

kilrah, 2016-12-05:
We still don't know what sequence of things we shoudl do to reproduce this. 
If this happened while deleting the model then we can't reproduce as it's not there anymore, we'd have needed a copy of the card BEFORE the problem.

3djc, 2016-12-05:
Or at least in the very same state on when the problem happened. When things liek that happen, you can remove the sd card, make a copy on your pc, then put it back and delete the model causing you problems. Like this, nothing to investigate

Githubbachelor, 2016-12-05:
Again this has noting to do with the deleted Model.5. Two different things!

The screen problem happen when I opened peace for peace each model to control your new NB360 in each model. And at the end of doing this happen that screen problem again. After switching off the TX and switch on everything again OK. But it happen everytime during opening any model and only 1 time a week maybe.

Githubbachelor, 2016-12-05:
The SD Card you get from me have had that problem.

schwabe, 2016-12-05:
@Githubbachelor yes but unless we can reproduce the problem it is hard to fix. To you deleting/not deleting a model might sound like no difference but it could be. E.g. the bug only might only happen if certain images are shown *together* on the screen. Deleting a model changes the order that behaviour.

Githubbachelor, 2016-12-05:
If I got any more problems about that thema I let you know. Maybe it is a problem between the SD Card and TX that anythink is limp behind, not quick enough???

Githubbachelor, 2016-12-31:
Until today the problem did not happen again. I think that was or is a problem if you are working with producing your models over hours. And when working it could be that the SD Card did not work right or too hot? Maybe a overflow of data during storing models or opening new? Or if you open for example ready made models after and after if you have more than 100 models for control all? In use of the horus that problem does not happen again since 1st. time during programming models.

kilrah, 2017-03-13:
Have you tried replacing your SD card at any point?

projectkk2glider, 2017-03-13:
This could be related #4589

Githubbachelor, 2017-03-13:
No I did not replace the SD Card only updated

kilrah, 2017-03-13:
Try to replace SD card. Have a feeling it could be casued by read errors.

Githubbachelor, 2017-03-13:
Yes that is what I think to. And now I know what you mean exactly. But it was not the card, I have now 3 differnt cards quickest on market. All the same problems. It The problem happen only if you open more than 14 Models after and after with quickly changing. Yesterday I want to control all my models for looking if I have set all positions right in the telemetry. So I open a model look into site telemetry and choose a other model for same procedure and after opening many models the picasso effect come. Than I have to switch off and on again and all is ok after. I think that is no big problem. That never happen if you use your model during a flight or with opened models

bsongis, 2017-03-25:
Feel free to reopen if needed

closed 2017-03-25 by bsongis

Githubbachelor, 2017-03-25:
Hallo Bertrand,
I think that that is a problem of some SD Cards. I made a test yesterday wit 4 different SD Cards all Class10.
For example also a problem with transmitting your eepe model dokument! Allways came half unknown models on Horus which were Ok before. If I retry and flash again in Companion, other models have lost data. 2 Cards ot the 4 were working perfect and to did not work. You cannot use Toshiba SD cards -that is a fact and anoname product was the second one.

Also it can be that many USB cables did not work with Horus! FRom 10 cables only one connect right!

So I think you can close the picasso effect with the known of bad CARDS! Or not compatible cables also.
Maybe you can give also info into the manual of open tx Horus and say to Helmut Renz.
Thanks.