opentx/opentx #4101: Taranis X9E  Pre-2.2 LUA Script Syntax Errors.

OldGazer, 2016-12-05:
Running custom scripts to support Teensy based MavLink to FrSky Smart Port telemetry for an APM 2.6 based on the work found in   [Scottflys' Teensy Telemetry Project](https://www.rcgroups.com/forums/member.php?u=560913).

Also running [Flight Deck 3.03](http://www.craftandtheoryllc.com/product/flightdeck-taranis-user-interface-for-ardupilot-arducopter-arduplane-pixhawk-frsky-telemetry-smartport-opentx/) from Craft and Theory for telemetry from a Pixhawk.

Radio is a Taranis X9E, receivers are FrSky X8Rs in D16 mode

The C&T interface uses "compiled" scripts which run just fine under OpenTX 2.2 RC7 and RC8.

The Teensy interface is fully functional.  The X9E detects all sensors.  Telemetry screens can be configured manually, but the custom scripts generate syntax errors.

3djc, 2016-12-05:
You should contact scripts authors so they update their scripts to 2.2 (if I understand correctly, you are trying to run 2.1 script with a 2.2 radio)

OldGazer, 2016-12-05:
You're understanding is correct.

Is there any way to find out what the differences are between 2.2 scripts and scripts for previous versions?

kilrah, 2016-12-05:
Documentation is still a work in progress.
https://opentx.gitbooks.io/opentx-2-2-lua-reference-guide/content/

projectkk2glider, 2016-12-06:
Issue for docs created https://github.com/opentx/lua-reference-guide/issues/25

closed 2016-12-09 by kilrah

projectkk2glider, 2016-12-11:
@OldGazer I made first rough document about the differences, see https://opentx.gitbooks.io/opentx-2-2-lua-reference-guide/content/part_v_-_converting_opentx_21_scripts.html.

Most likely your issue is with ``lcd.lock()` which should be removed from the script.

OldGazer, 2016-12-11:
Excellent!

I'll try this out and let you know the results.

OldGazer, 2016-12-11:
It wasn't  "lcd.loc().  It was file name length.

On a hunch I compared the Craft & Theory filename length to Scottflys Teensy telemetry. 

Not counting the extension, The C&T filename is 6 characters, Teensy filenames are 8 characters. 
On another hunch I shortened one of the Teensy file names to 6 characters, and darned if it didn't work.
There are some minor screen formatting adjustments that need to be made, so I'll turn that over to the original author.

Thank you so much for your time.