opentx/opentx #4102: Taranis X9E OpenTX 2.2 RC8 New Model Wizard "Hangs"

OldGazer, 2016-12-05:
The New Model Wizard "hangs" at the airplane icon and will not allow navigation to a different model type.
Only the setup for a new airplane can be completed using the wizard.

3djc, 2016-12-05:
Currently, only airplane is available.. If you don't want to create a plane, just use RTN

projectkk2glider, 2016-12-06:
I will reopen this to remind us to fix the wizard.

3djc, 2016-12-09:
Fixed, can be dl here : downloads.open-tx.org/2.2/nightly/sdcard/

closed 2016-12-09 by 3djc

OldGazer, 2016-12-09:
I'm confused.  This issue has been closed.  However, the files in the TARANIS archive have not been changed since 12/3/2016 and the wizard still "hangs."

3djc, 2016-12-09:
Give it another try. Also, as a general comment, when we close issues, it may not necessarily be accessible to you, it just mean that is done and approved, ready to ship in next version

OldGazer, 2016-12-09:
Thanks for the feedback.  I will keep this in mind...

I just tried the new script and it works.

Well Done

Thank you for taking the time to fix this..

PR #4105 (2016-12-08): Fix ROT encoder nav for wizards
Thix fixes #4102