opentx/opentx #4108: OpenTX Horus

joel-1, 2016-12-07:
Please double label/name fields from 4 characters to at least 8. When working with multiple items in the same family, the names get terse when limited to 4 characters. This impacts all items where a name/label can be defined. 8 or more characters would allow more meaningful names.

bsongis, 2016-12-07:
Not easy, 8 chars won't fit in all screens

kilrah, 2016-12-07:
The number of characters has indeed been chosen mostly based on available screen space. 

Allowing more would mean a complete UI redesign with a lot more submenus to display and edit things, that would be the end of the "everything at a glance" that is such a strong point of OpenTX.

joel-1, 2016-12-08:
I have not checked every screen, but on the ones I have used, there is room for more characters. On the inputs screen for example, I have my 3 character limited labels but other labels are longer and there's room for more. I think what really brought this to my attention was the only three characters for curves. So if 8 is difficult, would 6 be doable without a redo of the gui? I went through some screens and 6 seems OK and would help.

Thanks for all the great work on this!

Joel

[image]
[image]
[image]

guycnicholas, 2017-09-04:
It seems each display field could have a max length without limiting the length of the item itself. For example, I found the curve names to be painfully short, and I do a lot of my editing in the Companion app, where display length isn't really a problem.  So again, what if the names were limited to something like 16 characters, it would be so much easier to create/edit in Companion, then when that text gets displayed in various places on the radio, well, truncate to whatever you need to truncate to.

closed 2020-07-08 by stale[bot]