opentx/opentx #4113: FM7, FM8 not selectable from switch picklist (X9E/OTX 2.1.9)

RC-SOAR, 2016-12-09:
Minimal steps to reproduce:

1. Create new model in X9E (directly in transmitter)
2. Define FM7 (any name, any switch)
3. Create a special function, and attempt to set switch to 'FM7'

Problem: In the last step, it's not possible to select FM7 because the selection changes immediately to !FM7 in the UI. The setup performs as if switch is !FM7. 
Same problem applies with FM8. 
FM0 to FM6 work correctly

A user of one of my setups reported this. I don't have an X9E, however I am able to reproduce the problem using the Companion Sim, with profile set to X9E.  No problem with X9D. 

Same problem if setup is loaded from an .EEPE.

Transmitter: X9E
OpenTx version: 2.1.9

kilrah, 2016-12-09:
Yes it is a known problem that can't be solved on 2.1, the variable used to store the switch index is too small.

2.2 has it fixed.

closed 2016-12-09 by kilrah

bsongis, 2016-12-09:
Exactly 8bits are not enough to store all the switches values on X9E

RC-SOAR, 2016-12-09:
OK thanks. I can work around it by building an LS = !FM0 && !FM1 [...] && !FM6. Tests OK in the sim, although it needs a short (0.1 sec) delay to prevent a false positive at startup.