opentx/opentx #4114: wrong-freeze opentx 2.1.9 turnigy9x m128 (no processor green)

19alex86, 2016-12-09:
Hi everyone!!!! :)
I have a problem with my turnigy9x M128 processor "NO GREEN" ..
when I push the exit button repeatedly and not, not always, i have a reboot the system  or I have a freeze of the battery voltage. example. the voltage is 9.7 volts happens that suddenly drops to 8,6volt without the possibility of ricorreggerla from the menu and the only solution is to restart the radio.
What can you think? I done also this mod: "9x upgrade to ATmega128A3-AU + Reboot With Horiz Trims"  and indeed from right trim i haven't more the reboot problem
thanks to all in advance and sorry for my bad English!

schwabe, 2016-12-09:
Duplicate #3694. See also http://www.open-tx.org/2016/10/29/hk-9x-announcement

closed 2016-12-09 by schwabe

kilrah, 2016-12-09:
He said he did NOT have that processor.

But anyway this looks like a hardware problem, not our thing.

schwabe, 2016-12-09:
Oh sorry, I interpreted that as "No Green" m128 and for some reason thought no green was the chinese manufactor

19alex86, 2016-12-09:
i solved!!!!! :)
[image]

19alex86, 2016-12-09:
However, i want to signal a bug in the firmware 2.1.9 for turning 9x m128, in the section "channels" voice trim.. the issue is also in the simulator!

bsongis, 2016-12-09:
voice trim? would you explain more?

mhotar, 2016-12-10:
I had similar problem with my M128 chip i placed to the 9XR. I had this issue with Up,Down,Left,Right buttons. I placed a resistor in series with the switches. I sold this radio later, but I know new owner had the same problems with Menu/Exit buttons in some rare cases as well. It worked Ok with M64 but was strange with M128 http://openrcforums.com/forum/viewtopic.php?f=73&t=628&start=420#p37236