opentx/opentx #4121: lcd.drawnumber alignment issue

athertop, 2016-12-10:
In the companion 2.2 rc9 Taranis+ sim, the lua lcd.drawnumber function creates several issues if you want to right align numbers (I.e. if no LEFT is specified).

Using the following as a test script to demonstrate the issue:
`local function run(event)
  lcd.clear()
  lcd.drawNumber(30,1,420,MIDSIZE+PREC1)
  lcd.drawNumber(30,13,420,MIDSIZE+PREC2)
  lcd.drawNumber(30,25,1234,MIDSIZE)
  lcd.drawNumber(30,37,123,SMLSIZE+PREC1)
end
return {run=run}`

On OpenTx 2.1 this created nicely right aligned numbers:
[image]

but on 2.2 rc9:
[image]

Numbers align differently depending on changes in size and if no PREC is specified (for integers).
Also, in 2.2 they align further to the right than in 2.1 (I.e. not to the x coordinate specified)

3djc, 2016-12-14:
First of all, alignment default changed in 2.2, and is now LEFT. Specifying RIGHT produces a good result, exactly similar to what you have in 2.1, However, left alignment does seem to have the issues you mention, under investigation

athertop, 2016-12-14:
This is being produced by right alignment {the default for lcd.drawNumber) you can see the code used in my post. It clearly shows in the screenshot for 2.2 that indeed everything is not well with right alignment in 2.2.

3djc, 2016-12-14:
Please reread my comment : alignment default changed in 2.2, and is now LEFT

athertop, 2016-12-14:
Apologies. I have just tested specifying RIGHT in the bitmask and it does indeed align as per the previous functionality. Many thanks for clearing this up. 

So now I need to write additional code to format numbers based on the OTx version. In this case I will close this issue.

3djc, 2016-12-14:
As i said, while RIGHT is functioning ok, default LEFT is having an issue, so reopening it until this is fixed :)

athertop, 2016-12-14:
Again apologies. I don't really see the benefit of having the default behaviour changed in the first place. I can't help thinking effort would have been better spent developing a RIGHT alignment option for lcd.drawText instead - now this would be useful!

3djc, 2016-12-14:
We have obviously not done what you suggest. The whole lcd drawing handling was rewritten, and as a result, all drawing function are now LEFT defaulted

3djc, 2016-12-14:
Also, there is no need to develop alternative code for one or the other, you just add LEFT or RIGHT depending on what you want, not leaving it to default

3djc, 2016-12-14:
it should not

athertop, 2016-12-14:
I just tested it and I'm afraid it does! ..syntax error I mean.

3djc, 2016-12-14:
then RIGHT was not declared in 2.1 (can't recall). You'll have to be clever  then, but moving forward, both LEFT and RIGHT are declared

athertop, 2016-12-14:
No worries. Just to add, if I declare RIGHT in my code, then it no longer errors, and it does of course right align numbers, but not text unfortunately. Not a problem, I can live with adding this declaration and sticking with my existing formatting now I know that RIGHt works in 2.2, so happy days! Thanks again.

3djc, 2016-12-14:
I will prepare a patch to have LEFT working better in 2.2, but it does require a lot of testing

3djc, 2016-12-24:
LEFT and RIGHT now both work as they should

closed 2016-12-24 by 3djc