opentx/opentx #4122: Lua script file names limited to 6 chars in 2.2

mpaperno, 2016-12-10:
And I thought 8 was bad... ;-)  And since 2.1 all the telemetry scripts are in the same folder, which I think is good, but now the naming matters even more.  I assume this is an eeprom space issue (?), but it does seem kinda short (and a few people will surely wonder why some scripts aren't showing up in the lists anymore after upgrade).

Cheers,
-Max

3djc, 2016-12-10:
Indeed, we do not like this more than you do,  but it will have to stay  like that until next major eeprom change.  This is also a duplicate of #4117

closed 2016-12-10 by 3djc

mpaperno, 2016-12-10:
OK, thanks for quick reply, that's what I figured. And sorry for dupe (I did search but maybe not enough).

-Max

3djc, 2016-12-10:
No worries,  I have a feeling that this won't be the last issue on this :)