opentx/opentx #4127: Single expo curve in input option

OpenMik, 2016-12-11:
In input page, expo option is a symmetrical regulation of curve
Is possible to add a asymmetrical regulation similar the expo option in companion curve page?
For example Is very useful for adjust gas response

projectkk2glider, 2016-12-11:
Use two input lines and `Side` option set like this:
[image]

OpenMik, 2016-12-11:
Single Expo is like this
[image]
Is possible to obtain it by a curve, but  is more useful changing only a parameter in Input form 
[image]
Expo is already present in Input form, but with a symmetrical response

OpenMik, 2016-12-11:
[image]

OpenMik, 2016-12-12:
@projectkk2glider 
my request is different from how you get it
Other radio have this function and is very useful
currently to change the response of the engine need to edit more points individually on a curve
I suppose no more change on EEPROM is needed

Thank you

kilrah, 2016-12-12:
The function was actually added in companion as a workaround becasue it would have been too big a change in the firmware. Only 4 curve modes are possible using the current structure and they're all used, allowing for more would be a huge change for little audience.

closed 2016-12-12 by kilrah

projectkk2glider, 2016-12-13:
@OpenMik Try this workaround CH0
[image]
3 gives you the throttle response you want, expo amount is adjusted with GV1.

OpenMik, 2016-12-14:
@projectkk2glider 
Thank you for your Suggestion
I'll try it