opentx/opentx #4132: Mavlink not working with the 9X

PTDreamer, 2016-12-13:
The mavlink option doesn't seem to be working with the 9X.
Tried it with 57600 bps and nothing gets parsed.
One issue is that, according to the code, serial double speed is disabled which makes 57600bps greatly out of spec with the 16MHz 9X clock,
Also the serial driver seems messy and with no real difference compared to the also existing telemetry driver.
I can probably fix this even if being most used to working with ARM Cortex-M processors. If the maintainers want me to give it a shot just let me know if I should branch from main or next.

schwabe, 2016-12-13:
Most of AVR code hasn't changed much from master to next. There actually not many people working AVR code at the time, so feel free to go ahead. Also since nobody has actual mavlink equipment, the MAVLINK telemetry parsing was never ported to ARM radios either.

geofrancis, 2017-01-29:
It only works at 19200 or lower with the 9x and 9xr

bsongis, 2017-02-15:
@PTDreamer did your commits fix this issue?

geofrancis, 2017-02-15:
this fork works at 57600 and does frsky 
https://github.com/adriano65/opentx-2.0.14/releases

bsongis, 2017-02-16:
perhaps yes, but this is off topic :)

closed 2019-08-19 by kilrah