opentx/opentx #4133: _MAX_LFN is too long and not used "correctly"

mpaperno, 2016-12-13:
Continuing/splitting discussion from https://github.com/opentx/opentx/pull/4119#issuecomment-266289370 regarding maximum file name length with extension (_MAX_LFN) and maximum allowed full path length (undefined).

FatFs uses _MAX_LFN as the max. length for individual file or directory names (not their full path).  I think there is agreement that the current 255 is too long and a waste of RAM.

However a few places in the firmware code assume _MAX_LFN to be the maximum full path **and** file name. Eg. in radio_sdmanager (both versions) [here](https://github.com/opentx/opentx/blob/next/radio/src/gui/common/stdlcd/radio_sdmanager.cpp#L64-L73), [here](https://github.com/opentx/opentx/blob/next/radio/src/gui/common/stdlcd/radio_sdmanager.cpp#L286-L287), or in Theme [here](https://github.com/opentx/opentx/blob/next/radio/src/gui/480x272/theme.cpp#L53-L57).  There are some other places I've noticed in the code which could benefit from having a known set limit for the full path name instead of "guessing" or imposing conflicting limits (eg. [Clipboard](https://github.com/opentx/opentx/blob/next/radio/src/opentx.h#L1659-L1669) or [audio files](https://github.com/opentx/opentx/blob/next/radio/src/audio_arm.h#L68) or [Lua widgets](https://github.com/opentx/opentx/blob/next/radio/src/lua/widgets.cpp#L39)).

I propose to reduce _MAX_LFN and formalize a max. path name size limit separately from max. file name size.  This will reduce guesswork, make some string operations safer and possibly faster, and overall reduce runtime memory usage.

Just to throw some numbers out there, I'd suggest 47 or 63 chars for file name (with extension) and 47 or 63 for full path (w/out file name).  Those are all considerably longer than anything currently used by the system (sounds/themes/scripts/etc).  The extra length would be to accommodate user-provided paths and files like when using Lua I/O or SD-manager GUI operations.

If there is an agreement on doing this and on the max. sizes, I could submit patches.

Thanks,
-Max

lshems, 2016-12-14:
As said, for me 47 for filename seems fine, for the path I would suggest to stay on the safe side, 63.

Just my cup of tea.

bsongis, 2016-12-16:
I will vote for 47, this constant is already hardcoded in opentx.h (in the reusable buffer). It would make everything consistent. I would prefer that this change is only done in the 'next' branch though.

lshems, 2016-12-16:
to make things simpler, 47 is fine for me on path as well, if that's of any concern