opentx/opentx #4134: X9E OpenTX 2.2  Trainer Signal Lost, Trainer Signal Recovered alerts while making changes...

OldGazer, 2016-12-13:
Every time I edit model settings or radio settings I get repeated "Trainer Signal Lost" and "Trainer Signal Recovered alerts.

This happens regardless of the trainer port settings.
This also happens whether of not I have an aircraft powered up.
This appears to be caused by using the Enter button to scroll through a settings list.

I do not use the trainer port, so is there some way to turn the trainer function OFF?

3djc, 2016-12-14:
Cannot reproduce, maybe paste your eepe file ?

OldGazer, 2016-12-14:
 Q @   “! ˜"5 ¶": ±! w!# ù! m! U! !                                                                                                                                                                                                                C ¶" Mgearup¸
geardnÆloitr×althld²H—PñV( MÿÿÿÿÿÿÿÿHWXW4 VVV  JèëîîDÈ ôìTØ0ðüHÖ@üù‘¨NðüJåôì’TÈ  |÷ô“ôû“øî“ëüditr×althld²H—PñV(F L	Quad X4³HÁld‘KñôôÃdK÷ìýø³dKøîñììôûdKÿé. 
ñüûHëòûY‘NÁ2I‘‘L‘¡2I‘‘M‘¡2I‘‘K‘¡2I‘	 ÚŸ€ýâÄžÏþÉ¹	õæùôäûLõëÿà>B- ôÕ$+W‘H
‚dƒdd€d •
@’™ûûQÅ£en"ÁJJJJJ ‚å‘:å
‘$‘‘™ñ	"ÊñèìB„ÇëûôØðü òû£dKëüüûî£d Sd0JÿòüÄd8JûÿîÄd@	J÷ìýø
 VVOê‘ë‘ñ‘÷bò¡ ±w÷ô“ôû“øî“ëü˜ôì
íìrñùñ	"ÊñèìB  „DMfltdkSô’Lˆ’QfltdkMW cV VVV VVV  xˆ’{÷ô“ôû“øî“ëüsÿÿ¤ÿ0@ÿh L~d‘SdSdSdv‘NÁ2I‘ ‘L‘¡2I‘‘M‘¡2I‘‘K‘¡2I& lVV
 tdiaRú’“X’{÷ô“ôû“øî“ëü”ôì
wôì’TÈ? L~d‘KñôôÃdK÷ìýø³dKøîñììôûdKÿé øñûò÷è’§Pcsimyd‘SdSdSdSd Kûÿî |Yx‘NÁ2K‘L‘¡2K‘M‘¡2K‘K‘¡2K‘d‘¡2< c ‘d‘©2ñíìôîó‘sH ÿVV VVV8 l÷ôûñ”ôûêIøîñIëüüI
ñüûáNòü
ñüöûîùÑ ûÿîf‘NÁ2K‘L‘¡2K‘M‘¡2K‘K‘¡2K‘O‘¡2 I‘‘d‘¡2I‘‘c‘¡2K‘U‘¡2I"   IQuadã`	’°‘Nld‘KñôôÃdK÷ìýø³dKøîñììôûdK ÿéÓd Kôì
ñüûn÷ôûñ”ôûêIøîñIëüüI$ ãàå•ÿýûî¸Qav250ÃHÁO\d‘KñôôÃdK÷ìýø³dKøîñìì% ôûdKÿéÓd Kôì
ñüû“d(Kñíìôîó v‘NÁ2I‘‘L‘¡2I‘‘M‘¡2I‘‘K‘¡2I‘‘a‘¨2ôì
ñüû¡‘D VVV( VVVV) VVV  y‘|÷ô“ôû“øî“ëüdV, ÷ì]Á¡ÿ•‘:Ùñ	"ÊñèìB„¸ðüJåôì’TÈ3 VVV  ‚^“øî“ëüdy‘÷  JâòIÿ{lêIøîñIëüüI
ñüûáNòü
ñüöûîùÑ/ ‘a‘¡2Iùstblzmd©loi0 trmd¨	althld²HPôPô(IacroYÿÿ’{^1 XWV2 VVV VMîóÅ
íìÿs	sfthudRsfttxtRsfH Vy‘|÷ô“ôû“øîE VVMfltd6 !#îñæHexaãp’8‘i¡d‘K÷ôûîñò“dKôûêÿìñîdKòù÷7 òû£dKëüüûî£d Sd0JÿòüÄd8JûÿîÄd@	J÷ìýø> l÷ôûñ”ôûêIøîñIëüüI
ñüûáNòü
ñüöûîùÑ9 –÷ó÷ìo‘NÁ2I‘‘L‘¡2I‘‘M‘¡2I‘‘K‘¡2I‘‘a‘¡2L I‘‘d‘¡2I‘‘c‘¡2K‘U‘¡2I; 	÷óëôÿìñî·Pcsimv€¡d‘KñôôÃdK÷ìýø³dKøîñììôûdK ÿéÓd d¢€ôûÿîo“÷ôJôûH“øîH“ëüJ= VVVA VVV  –÷ó÷ìo‘NÁ2I‘‘L‘¡2I‘‘M‘¡2I‘‘K‘¡2I‘‘a‘¡2@ h‘NÁ2I‘‘L‘¡2I‘‘M‘¡2I‘‘K‘¡2I' lB Vxð~÷ô“ôû“øî“ëü”ûÿî  _xˆò¡~÷ô“ôû“øî“ëütÿ¤þÀg !#îñæHexaãp’8‘i¡d‘K÷ôûîñò“dKôûêÿìñîdKòù÷ VVy‘* kSô’Ñ€â€’QfltdkMW÷ô“ôû“øî“ëü”ôì
Ãòü˜ûÿîI h‘NÁ2I‘‘L‘¡2I‘‘M‘¡2I‘‘KK MÿÿÿÿÿÿÿÿHWXW  “ëüdñèìB„·ççõ‘6åôì’J ‘¡2IZH+ ÿVVVM VVVG Mgearup¸
geardnÆloitr×althld²H—PñV(N VVMfltdO kSô’Ñ€â€’QfltdkMW÷ô“ôû“øî“ëü”ôì
Ãòü˜ûÿîP ÷ì]Á¡ÿ•‘:Ùñ	"ÊñèìB„¸ðüJåôì’TÈ  ‚^                                                  R                                                               S                                                            [... cut ...]                                                   õ                                                              ö                                                              ÷                                                              ø                                                              ù                                                              ú                                                              û                                                              ü

OldGazer, 2016-12-14:
I'm going to try posting a video on YouTube later this morning...

OldGazer, 2016-12-14:
Video up loaded to YouTube:  https://youtu.be/wpamZM-_rwo

3djc, 2016-12-14:
Your screen is blinking at the same time, hardware issue ?

OldGazer, 2016-12-14:
I think you nailed it.

On a hunch I played with the Haptic settings.

If I turn the Strength and Length down to the mid-point (I had then both set to max), I do not get the alerts and the screen does not flicker, but...  I can't feel the vibrations...  So, I set the Haptic to Quiet and tweaked the Sound settings to compensate for my hearing impairment..

So far, there are no more alerts and the Haptic still buzzes for power on and power off.

I think I can live with this..

I'll let the folks on RCG know..

kilrah, 2016-12-14:
> Your screen is blinking at the same time, hardware issue ?

Can't watch the video from where I am but that sounds like the "blink backlight on alert" feature ("Alarm" under Backlight in radio settings).

OldGazer, 2016-12-14:
I'll check that out...

Thanks...

OldGazer, 2016-12-14:
I unchecked  Alarm under Backlight and turned Haptic back on.

I created a new model and messed with the settings.  The Trainer alert sounded once instead of multiple times.

I went to an existing model and did the same thing.  The alert never sounded.

I'll check this a couple more times and report back.

OldGazer, 2016-12-23:
I have not had this issue for several days now, so I consider it to be resolved.

Many thanks to every one on the dev team.

Cliff

closed 2016-12-23 by OldGazer

projectkk2glider, 2016-12-23:
What bothers me is that we never actually found out what was the reason for this issue. Perhaps there is still a software bug lurking somewhere.

OldGazer, 2016-12-23:
I've been thinking the same thing, but for the life of me I cannot reproduce the behavior...

Cpt-Huck, 2022-06-18:
FlySky Nirvana 14 (NV14) - EdgeTX 2.7.1
just had this when writing an Backup via EdgeTX Configurator to my Radio.
If I go to "Model Setup" - scroll all the way down - and on "Trainer" - "Mode" - "Master Jack" is the default I guess ?!
Gives me the same problem you have "Trainer Signal Lost / Recovered". (Random ?! and when navigating through the menu ?!)
"Master/Multi" = no problems. But I just turned it "OFF" since im not using it. = Problem solved / fixed with this Setting.