opentx/opentx #4135: Feature request: bundle eepe + bmp + wavs + lua scripts for import/export

R5fan, 2016-12-14:
I have a small request: in order to make it easier to share model settings and lua scripts with other users, make a way in companion to bundle an aircraft eepe along with selected (or actually used) WAVs, its BMP and LUA scripts (and folders) in to a single file or container for easy import/export and sharing.

Also, in companion there doesnt seem to be a straight forward way to export and import a single model as is. I tend to create a new eepe, copy the aircraft there, then export as eepe.  On the radio, you can backup a single model. If that functionality exists in companion, Ive not found it (saving as bin then restoring doesnt seem to work?).

lshems, 2016-12-28:
In Companion, start the simulation, and save the model using the radio menu as you would do on the radio. That's what a sim intends to do, simulate the radio.

This backup file is MUCH smaller then an eepe, and doesn't mess up transmitter or companion settings when opened from within companion by someone else.

It should be the default format to exchange models, instead of EPEE, with all the calibration stuff and so in it.

It would be a nice to have to save a single model from compnion directly in this model backup format,, but since it can already be done in some (not convenient) way, normally it will not end up high on the to-do list here. But you'll never know.

Good luck.

projectkk2glider, 2016-12-28:
The save of the individual model is pretty straightforward.

 The other idea about complete bundle needs more thought and discussion.

lshems, 2016-12-29:
Building the package could be done in lua. In lua you van Check what scripts are used on mixer, SF etc. Only the TELEM scripts are not available to set from lua yet.

closed 2019-11-15 by stale[bot]