opentx/opentx #4137: Audio Noise  - Horus

robthomson, 2016-12-15:
When using opentx on the horus - there is a persistant background hiss

This is not present when running the frsky os.

Possibile solution... turn the amp off when the radio is not playing sound?

projectkk2glider, 2016-12-28:
Turning off the amplifier might even save some power, but two things to check:
 * verify that amp on/off does not produce noise
 * determine how quickly amp turns on, ie is it possible to turn it on when we have something to play and then send ADC data immediately?

Xerravon, 2019-10-30:
I get this same problem but with the crossfire TX installed. it is very loud when crossfire gets power. it is very loud when set to 868mhz but quieter when set to 915mhz or when the unit connects with the rx. It is very quite once the rx connects.
I also noticed if I set the crossfire from "normal" to "Forced" in telem setting it changes from a constant buzz sound to a pulsed sound...
hope this helps. I also sent TBS the same info.
thanks
John

bsongis, 2019-11-07:
Done by @raphaelcoeffic on X10

closed 2019-11-07 by bsongis