opentx/opentx #4143: X7 misc issues

kilrah, 2016-12-17:
- [ ] Using source popup doesn't work and caused reboot on certain commits (when pressing enter, or when scrolling past displayed items)
- [ ] Missing unexpectedShutdown flag on main view 
- [ ] Screenshot SF outputs valid image but with garbage
- [ ] Can't scroll in text view
- [ ] SH not defined as momentary
- [ ] TMR3 not displayed in stats (maybe not necessary...)

bsongis, 2016-12-17:
For SH, would you do a fresh erase of your EEPROM to be sure?

bsongis, 2016-12-17:
What do we do with TRM3 then?

kilrah, 2016-12-17:
Let's forget about TMR3 for now.

I confirm SH even after a reset, selecting it as a switch som a field and using autoswitch does not work as toggle. Also it seems S1/S2 are set as "pot without detent" as the center position is not being calibrated

bsongis, 2016-12-17:
Which script did you try? Is it reproducible with the snake script?

kilrah, 2016-12-18:
I don't remember which script it was - I now can't seem to get any standalone script to run including snake (and existing Taranis telemetry scripts like BattCheck cause a reboot)

3djc, 2016-12-18:
They would be drawing outside screen given X7 size, no ?

kilrah, 2016-12-18:
Yes but shouldn't we have protection for that?

3djc, 2016-12-18:
I think it would be very wise yes :)

bsongis, 2016-12-18:
@kilrah would you only try the scripts from the official SD? Here the snake is perfectly OK and exits properly with [EXIT Long]

kilrah, 2016-12-18:
OK the X7 snake is OK indeed!

bsongis, 2016-12-19:
Sure, which script would be a good candidate for those tests (and crashes now)?

kilrah, 2016-12-19:
For example this, run as telemetry script causes a reboot when the screen is displayed

```
local function run_func(event)

  lcd.clear()

  lcd.drawText(10,10,"Hello World \128 \129", MIDSIZE)

end

return { run=run_func }

```

closed 2016-12-19 by kilrah