opentx/opentx #4156: Little error when using lua string.format and lcd.drawText

pau33, 2016-12-22:
I try to show the preceeding +- on the telemetry screen.
It works only for the smaler fonts:

    lcd.drawText(0,0,string.format("%+04i",23),SMLSIZE)
    lcd.drawText(0,10,string.format("%+04i",23))
    lcd.drawText(0,20,string.format("%+04i",23),MIDSIZE)
    lcd.drawText(0,40,string.format("%+04i",23),DBLSIZE)
    lcd.drawText(50,20,string.format("%+04i",23),XXLSIZE)

Result:
[image]

It is the same result, if I use %i or %d
Taranis Plus  OTX2.1.9

Thank you in advance.

projectkk2glider, 2016-12-22:
Both DBLSIZE and XXLSIZE fonts don't have `+` character at all (except on Horus). Should we add them or not?

kilrah, 2016-12-22:
No reason not to I guess

3djc, 2016-12-22:
Just let me finish the drawNumber PR before changing it on next please (or at least please start from DBLSIZE in my branch)

pau33, 2016-12-22:
Thank you for the fast response.

There are more characters missing in the large fonts, for example < or > and some more. Others appear differnet on the panel than on the keybord.

May be not all characters must be implemented, but the implemented ones should be described in the documentation.

pau33, 2016-12-23:
May be an appropriate subset of characters could be:
"=+-_.:,;<>|^*#()!?"

Some work already.

kilrah, 2016-12-23:
You need to look at the standard size font file - that's what can be implemented.
DBLSIZE should have all supported chars, XXLSIZE has only numbers (storing the whole font is starting to become ridiculously huge in memory requirements).

pau33, 2016-12-23:
Are you shure?
With     lcd.drawText(0,20,"=+-_.:,;<>|^*#()!?",DBLSIZE)  I get:

[image]

For the XXLSIZE I agree, only number an +- should be sufficient.

kilrah, 2016-12-23:
Well it is in the font file, but something else could be broken. 

https://github.com/opentx/opentx/blob/next/radio/src/fonts/std/font_10x14.png

projectkk2glider, 2016-12-23:
That font file si never used, the used font is https://github.com/opentx/opentx/blob/next/radio/src/fonts/std/font_10x14_compressed.png

lshems, 2016-12-26:
Would it be logical to stick to  the characters used in the string.byte and string.char functions?

pau33, 2016-12-26:
In my opinion that is a good idea and logical. That would show the ASCII characters from 32-126 correctly on the screen (string.char).
If memory size is a problem at the big fonts, it could be less (XXLSize numbers and +-).

closed 2019-11-15 by stale[bot]