opentx/opentx #4159: Open TX 2.2 - Simulator "PF7 - Reload Lua Scripts"

rcbebo82, 2016-12-23:
Im currently programming a widget. In the simulator the option Reload LUA Scripts does not work.
I edit the script in the background click on "PF7 - Reload LUA" scripts but the script does not seem to be loaded in the new version. I have to close an start the simu again to make it working.

mpaperno, 2016-12-23:
Do you mean that specifically the Lua widgets on Horus aren't reloaded?  I can confirm this.  The MIXES scrips seem to reload OK (and telemetry scripts on Taranis).

mpaperno, 2016-12-23:
They're also not reloaded when Custom Scripts (MIXES) are updated.  I believe the widgets are only loaded at startup.  This may be more of a feature request than bug per-se.  Definitely would be nice though.

projectkk2glider, 2016-12-23:
That is because I separated the two Lua environments and forgot about this feature. But on Taranis it should be the same.

rcbebo82, 2016-12-23:
For sure only widget scripts. It costs a lot of time when you develop a widget and have to restart the sim each time. Would be nice if the button also forces the widgets to reload this would speed up development per se. If the widget contains telemetry sources you also have to startup telemetry each time and set it to your needs.

projectkk2glider, 2016-12-23:
It is clear now where the problem is. This will be fixed.

projectkk2glider, 2016-12-23:
This will be harder than I first thought. Not only has the Lua environment be reloaded, the whole system of registering themes and widgets, creating widgets, etc must be recreated. Basically it will look like a new model was selected and loaded. 77

rcbebo82, 2016-12-23:
Maybe "Reset Flight Data" could do this also?

3djc, 2017-03-06:
We have gone as far as we can on this by providing a reload radio that doesn't require to restart the simu software. Closing this as a result

closed 2017-03-06 by 3djc