opentx/opentx #4162: Question LUA startup order

pau33, 2016-12-24:
I have an extended mix script an now I added a telemetry script to show some results during flight.
Both scripts communicate via global variables, the telemetry script is the master.

To save some computing power I wanted to perform an arithmetic calculation to one of the global variables (from the mix script) in the init-function of the telemetry script and save the result in a local variable in the telemetry script.

All works fine in the companion-simulation (OTX2.1.9, Taranis Plus).
But when I flash it to the radio, I receive an error in the telemetry-script, when performing the formula in the init-section.

I assume, that the start order of the LUA-scripts is different in simulation and reality and the global variable is NAN at startup time of the telemetry script.

Now the question: Is there a posibility to predict or better to control the startup-order of the LUA-scrips? Or can there be another problem, I do not think about yet?

Thank you for your answers.

pau33, 2016-12-25:
The simple solution should be to define the global values in all definition sections of all participating scripts (as in the example script). But of course the underlying question of lua startup order is still interesting, especially for client server interactions.

In my case this solution did not work may be because of memory problems in the radio (in the simulation everything works fine). I now define the global variables (10 in total) in the telemetry-script. I can use them in the mix script, but when I try to define any global variable in the mix-script it crashes on the radio (not in the simulation). Interesting is, when I define additional local variables in the mix script, that works. So the solution is not fully satisfying for me yet.
Is there a way to find out the memory usage of variables and scripts  in the radio, not using special debuggers? Or are there other debugging techniques for mix-scripts on the radio?

mpaperno, 2016-12-25:
I don't really see how the scripts would load in a different order on the radio vs. sim.  They're all loaded in the order specified by the user -- first the custom/model scripts (MIXES), then functions, then telemetry screens.  The code is [here](https://github.com/opentx/opentx/blob/master/radio/src/lua/interface.cpp#L437)... there's not much room for "variation" that I can see.

Quite possibly you're running out of memory on the radio.  The only way I know of determining for sure is to run a debug version of the firmware on the radio and use a serial console to read the actual error happening.  

There are some features on the way to improve the memory usage issue (see #3318 & #4119). In the meantime you could try "optimizing" the code by using shorter variable names, reducing complexity, or perhaps other tricks.  Splitting functionality up into smaller files may also help.

-Max

mpaperno, 2016-12-25:
PS. I also use mixer scripts which depend on each other and telemetry scripts which depend on those, and never ran into an issue with loading order on sim or radio.

pau33, 2016-12-25:
Thank you for the response. Is see, the startup order is fixed.
I am now quite shure, that I run into memory problems. The mix scripts now behaves instable, also I apply only minor changes. Meanwhile I commented out a part in the mix script, I used for debugging on the radio and it seems to start. I have to try during the next flights.
Features to improve the memory are welcome :)

closed 2016-12-25 by mpaperno