opentx/opentx #4163: Negative GV's as Input Weights no longer works in 2.2.0N360

roycegb, 2016-12-24:
I can no longer use negative Global Values for input weights as of R360. The resulting weights pass only very small or no stimuli.  Positive GV's work correctly.  Negative GV's may have worked in the September releases of 2.2 but certainly work in 2.1.9.

kilrah, 2016-12-24:
What about some useful details, like on what radio?

roycegb, 2016-12-24:
Andre,

Sorry, Taranis X9D Plus.  The failure shows up on the Tx or on the Companion 2.2 simulations.  If you look at the resulting Curve, it is squashed, probably a math error.

The inputs are shown in the image, (-GV1) weights causes the problem only in 2.2.0.  (GV1) weights work fine in 2.2.0.  

The receiver screen shots show that the inverse Curve seems to be calculated wrong. It should be the same basic extent but opposite slope.   Thw (throw) is GV1 on the two screen shots.

Thanks,

Royce

projectkk2glider, 2016-12-25:
I can't reproduce
[image]

@roycegb your pictures are not showing.

kilrah, 2016-12-25:
Maybe he means when you select -GV1 as weight?

bsongis, 2016-12-25:
-GV1 as mixer weight works perfectly
testing -GV1 as input weight now ...

bsongis, 2016-12-25:
[image]

roycegb, 2016-12-26:
Bernard,

I had built my original model files in version 2.1.  I brought those over into version 2.2.  Apparently the format translator makes some sort of error.  If I generate totally new versions of each model within 2.2 they seem to work correctly.  

I think i read something about translator problems in the OpenTX discussion under the S6R topic.

Thanks for your help.

Royce

schwabe, 2016-12-26:
do you still have a backup of your 2.1 files? Reproducing the error would allow us to fix it.

bsongis, 2016-12-26:
this is very possible I forgot something in model translations, I need a backup to reproduce!

roycegb, 2016-12-26:
Bertrand,

Here are the .eepe files for version 2.1 (works) and then version 2.2 (doesn’t work).  You can see the problem in Companion 2.2 with the version 2.2 file or replicate the conversion error by opening the version 2.1 file with Companion 2.2.

I am using Companion Version 2.2.0N360, Dec 4 2016 on a MacBook Pro running OSX version 10.11.6.  I have not tried running on Windows or Linux.

Royce

schwabe, 2016-12-26:
The attachments you probably made to your email did not get through to github. And just to be clear you made the conversion with Companion and not with the radio itself?

kilrah, 2016-12-26:
Attachements are not supported when replying via e-mail. Need to use the web interface.

roycegb, 2016-12-27:
Arne,

Yes, I made the conversions in the latest Companion and not the radio.

Royce

bsongis, 2016-12-28:
Conversion tested on Companion. Again everything fine. Please reopen with the attachments if needed.

closed 2016-12-28 by bsongis