opentx/opentx #4166: Final CH outputs loging + LSW

janekx, 2016-12-27:
For some purposes  there will be handful to have posibility to log the final CH levels and not only imputs.

Today there is possible only log SW, gimbal but not CH levels and Logical SW.

projectkk2glider, 2016-12-27:
Logical Switches are already logged in 2.2, the column is named `LSW` and format is 64bit hexadecimal number, bit 0 is LS1, bit 1 is LS2, etc... The decoding of this in Companion log viewer is not implemented yet, not sure how to do this in an visually pleasing way.

Short example log:
```
Date,Time,SWR,RSSI(mW),RxBt(V),Tmp2(@C),VSpd(m/s),Alt(m),Rud,Ele,Thr,Ail,S1,S2,S3,LS,RS,SA,SB,SC,SD,SE,SF,SG,SH,LSW,TxBat(V)
2008-10-02,17:36:45.570,30,75,5.18,0.0,0.0,0.0,0,0,0,0,0,0,0,0,0,0,0,-1,0,0,1,0,-1,0x0000000000A20000,0.2
2008-10-02,17:36:46.070,30,75,5.18,0.0,0.0,0.0,0,0,0,0,0,0,0,0,0,0,0,-1,0,0,1,0,-1,0x0000000000B20000,6.0
2008-10-02,17:36:46.580,30,75,5.18,0.0,0.0,0.0,0,0,0,0,0,0,0,0,0,0,0,-1,0,0,1,0,-1,0x0000000000B20000,6.0
2008-10-02,17:36:47.090,30,75,5.18,0.0,0.0,0.0,0,0,0,0,0,0,0,0,0,0,0,-1,0,0,1,0,-1,0x0000000000B20000,6.0
```

The `0x0000000000A20000` is one example of LSW state.

Logging of actual channel outputs is not implemented. This would increase the size of the logs a lot (which is also a performance issue, not a capacity issue).

kilrah, 2016-12-27:
To do it properly we should really be able to enable/disable the different groups. Someone who wanted to log the channels would likely be able to disable both the analogs and switches to more-than-compensate for it.

projectkk2glider, 2016-12-27:
Makes sense, also I think nobody wants to log all 32 channels, there should be an option to choose which channels to log (at least in groups of 1-8, 9-16, or something like that).

bsongis, 2016-12-28:
A popup menu in the outputs screen? Plus a small icon as @mhotar designed for GVARS popup?

kilrah, 2016-12-28:
Why not... I'd have thought of a "Logging" submenu on the telemetry page but that might be better

VancouverUmbrella, 2020-02-27:
I'd vote to keep this issue open. Channel logging could come in handy.

lshems, 2020-03-06:
There is in the mean time a LUA workaround. Works in most cases. 
https://www.justfly.solutions/index.php/full-repository?view=product&id=110:logchannels

It uses the mixer values and the output settings to calculate the output and send it as a sensor.

trims, min, max, ppmoffset, trimtype are standard supported
override channels need to be listed in a table manually (BUG in opentx on function value in Special Function LUA API), but once done is also accounted for.

No support (yet) for curves and Gvar on trims

VancouverUmbrella, 2020-03-06:
Yes, works quite nicely.

Just need to remember to delete the first generated log file so columns will be created properly -- as noted in script file.

lshems, 2020-03-06:
Problem is that my repository is not commonly found yet.
Spread the word :). and yes, the repo is free because it attracts users to the paid part if the site as well. Nothing fancy, but nothing to hide neither.

git2323, 2020-09-08:
would be nice to have this implemented

closed 2021-06-18 by stale[bot]