opentx/opentx #4174: RSSI scaling is different from OTX 2.1?

RipperDrone, 2016-12-30:
Flashed OpenTX 2.2 and Betaflight 3.1 to try out live PID tuning. I noticed that the RSSI values displayed vai FrSky telemetry have changed significantly, Taranis reads RSSI = 1 when being close.

I bumped the scaling up by 100, and now the close-by RSSI is around 30. Since Taranis also has a warning level, I'm not sure what to think about it: Have you changed the scaling for a good reason? Or is it simply a bug?

bsongis, 2016-12-30:
Are you sure? If yes this is not an intentional change

RipperDrone, 2016-12-30:
Yes, I verified it with 2 quadcopters, both of them having shown 100dB based scaling before. Since OTX2.2, both of them show RSSI=1 when I'm close. I applied a manual factor of 100, but still this seems different from what it was before (shows RSSI = 30 then when I'm close to the copter, was close to 100 before). Therefore, manual scaling needs to be 300? Strange...

RipperDrone, 2016-12-30:
Just to be complete & accurate in describing my test environment:

D4R bound to Taranis OTX2.2 (mid Nov version), using FrSky telemetry. All other telemetry values seem correct (radio voltage, copter voltage, heading, ...). It's just the RSSI that has changed. I did a new sensor search - all existing sensors confirmed and no new sensors found. So it looks like a bug, I guess?

3djc, 2016-12-31:
Just tested X9E/OTX2.2N360 with X6R, showing 96dB when 1m distance about, will try d4r later

3djc, 2016-12-31:
Tested with D4R-II same normal values

bsongis, 2016-12-31:
Did you notice this problem in previous 2.2 RC? It's really strange!

3djc, 2016-12-31:
The only thing I can think of is a migration issue, because I've used RSSI on all RC (and before) on my taranis (but no migration)

3djc, 2016-12-31:
@RipperDrone Have you deleted the sensors before search ? Or just launched discover new sensor while leaving existing in place ?

kilrah, 2016-12-31:
And what if you create a new model memory?

RipperDrone, 2016-12-31:
Thanks for all your valuable inputs.

- I have not deleted and then searched sensors again. Will try.
- migration: Yes, there has been a glitch that OTX always showed a warning in my Taranis after having upgraded from 2.1 to 2.2 RC8, obviously a bad version/sub version count in the ID file. I manually corrected the version count (from something like 203 to 204) to override the warning on Taranis startup.

Any advice how to upgrade to the RC8 correctly, without any version warning showing everyone I switch the RC in?

Furthermore, since I have 10 quite complex model definitions, I reloaded all models into RC8. Maybe this migration has overwritten some of the sensors scaling code in a bad way?

Anyway, will try the sensors new discover first... ☺

RipperDrone, 2016-12-31:
@3djc You made my day - delete and re-discover did the trick. Correct scaling/dB unit recovered! Must be a migration issue. Hopefully covered by the final "update and still keep models" routine. Thank you to all of of you - and a Happy New Year! I will close this issue then :-)

RipperDrone, 2016-12-31:
Did some follow-up research: It's obviously still about units / ratio / offset migration: When I set the correct parameters on my Taranis and read them back into OTX, units of RSSI are changed from dB to RPM, and ratio/offset parameters are changed as well. There seems to be a mismatch in the data structure.

Do I need to use specific Nightly Builds of OTX on my computer visa same Nightly Build version of a firmware falsh to Taranis to fix this?

RipperDrone, 2016-12-31:
`Here is what I could reproduce: 

- verified that I have OTX 2.2/N360 installed on my Win 10 computer
- re-formatted my Taranis X9D+ SD card
- copied RC8 SD card content on SD card
- started up Taranis: It says SD card warning, Expected version 2.2V0003 (visa SDcard version file having v0004)
- looks like the EEPROM file on my Taranis has not been updated by OTX N360. How can I do it manually?

RipperDrone, 2016-12-31:
It has to do with mismatch btw Firmware and OTX 2.2.0. Flashing did not work in the beginning because splash screen was missing. Selecting the FW splash screen let me finish the firmware update now. Will do further investigation if migrations are consistent now...

RipperDrone, 2016-12-31:
Have managed to repeat the bug as a whole now:

1. Installed OTX N360 on my PC
2. Flashed N360 firmware (v0004) onto Taranis X9D+
3. changed RSSI telemetry units setting on Taranis (unit = dB)
4. loaded settings & models from taranis in to OTX on PC

Result: Units of RSSI have changed from dB to RPM, and switches like blades/multipliers/log have been checked automatically. Seems to be a bug in the routine handling the read from RC

Is this a known bug or a new one?

[image]

RipperDrone, 2016-12-31:
Looks like related to issue #4031. It should have been fixed by now, but it seems it isn't.

@Githubbachelor, @kilrah , @3djc: You contributed to the issue #4031 thread in Nov this year...

3djc, 2016-12-31:
#4031 did not involve companion at all, just a german language issue, that as indeed been fixed

RipperDrone, 2016-12-31:
I see - so this is a new bug. Just sounded similar - jumping from dB unit to RPM...

Githubbachelor, 2016-12-31:
@RipperDrone @3djc 

That Problem is still cleared and fixed with 360. I have German Language. I changed RPM into db in the editor of Sensors and all is done. If you produce new model now, all is right!

RipperDrone, 2016-12-31:
@Githubbachelor 

Yes, correcting the RSSI unit on the Taranis is fixed. However, reading from Taranis into Companion still messes up the RSSI units, as described before.

RipperDrone, 2016-12-31:
For those interested in a workaround which I've just tested - can be used until issue is fixed: 

When NOT entering a unit along with RSSI (dB), but just setting the unit field to 'raw(-)', everything works ok: Reading data into Companion, then writing it back to Taranis yields consostent data.

3djc, 2017-01-02:
Fixed in next nightly

PR #4181 (2017-01-02): Add missing units
Companion was not updated when mWatt and Rad where added

This fixes #4174
closed 2017-01-02 by 3djc