opentx/opentx #4177: Invalid switch position options visiable when changing between 3pos and 2pos switches

robthomson, 2016-12-30:
Best to explain this issue by example.

On my horus I have physically changed two of my 3pos switches into 2pos switches.

On the config page in system I set the switches to be 2POS.

All good.

But...

If you do things like scroll through list items that show switches. (e.g.  failsafe)

one will note that the 'dormant' switch position still comes up in the list.

This does not cause any problems - but is from a design / gui point of view an error.

An issue does further come to mind however when people do these sort of mods.  Switch positions become quite important.  in my case due to the wiring on my two switches I changed - one generates +100 in one position - the other -100 for the same direction.

This is correctable at a hardware level; but it does strike me that the calibration routine could me more intellegent when it comes to handling switch positions.

e.g.:   set all switch positions to 'forward'.  detect.  - set all switch positions to rearward - detect etc..

Certainly not a critical issue; but would be a good option for those whole like to change items on the radio!

bsongis, 2016-12-30:
This should be implemented (hide non existing switches positions). Where do you see it? Always? I am confused with "failsafe", which screen, special functions? I just tested, no issue there.

robthomson, 2016-12-31:
I see it on the horus.. 

On the switch warning page on each model config. 

Not checked anywhere else yet!

bsongis, 2016-12-31:
I didn't check the switch warning page yet

bsongis, 2016-12-31:
Bug confirmed, in Switches warning line.
While on this, I think that up/down arrow at the right of the switch name is 1 or 2 px too much at the right

3djc, 2016-12-31:
I'll move the arrow a bit to the left

commit b77332e7 (2017-01-04): [Horus] Fixes #4177
closed 2017-01-04 by bsongis