opentx/opentx #4178: Feature Proposal for Settings: Paste to All Models

RipperDrone, 2016-12-31:
I know this might have some risks / drawbacks for the noob users, but many times I have come across situations where I'd like to add a function for all models (e.g. low voltage battery warning set up by a number of switches and wait time triggers...). In Global Functions, a lot of the ingredients are still missing to have the full set of functions available. On my Taranis, there are 10+ models, so pasting into one model at a time is a tedious task...

Therefore, a simple feature extension could be a "Paste to All Models" function after copying a line from the inputs/outputs/special functions etc. tabs. This reliefs us from having to paste into all models individually.

Maybe an additional warning dialog makes sense: "Are you sure? This will overwrite any existing content in the same line in all your models!" - Yes / Cancel

3djc, 2016-12-31:
While a nice in theory idea, there are plenty of practical reason why this can't really be achieved. If those things are not in globals, it is because of the same reason : you cannot guarantee that all your models will have the required 'items' for it to work

RipperDrone, 2016-12-31:
How would you set your Taranis for a decent low voltage warning over all models? Maybe I'm missing an elegant way to do so. Currently, I have to replicate these LS settings onto all models individually, let alone the special functions which have to be adjusted on a one by one basis as well:

[image]

3djc, 2016-12-31:
Low voltage great example : some of my drones use vbat through cleanflight telemetry (so you get it as VFAS), some use X6R analog in (so you get it as A2), most use flvss (so I setup alarm on lowest cell, not total), some use 12S so two flvss, and the 'petrol' engine based use RxBt ...

Their simply isn't a one size fits all that will work well (and safely) for everybody. And that's just the tip of the iceberg, technically, that would be even worse

RipperDrone, 2016-12-31:
So could there then be at minimum a way of copying whole tabs of Logical Switches, Special Functions, Telemetry etc. rather than having to go through line by line and copy/paste each line individually.

With such a feature, it's still up to the user to deliberately decide which tabs to copy into which models (like many of us have a bunch of similar / identical models set up, they can then be worked over as a cluster more easily).

From a coding standpoint, I don't see it's too difficult to handle copy/paste of whole tabs in addition to copy/pasting individual lines of each tab only.

What do you think?

kilrah, 2016-12-31:
There is already an issue about copying an entire tab, that's indeed open for consideration. 

But usually in the cases where you want to do that and you have so many settings in common you may be better off starting the new model as a copy of an existing one and changing what needs changing, instead of creating a new blank one and copying everything that's identical.

RipperDrone, 2016-12-31:
Yes, that's an option. However, I've met many use cases where I wanted to extend a cluster of models by one feature, requiring some LS / SF defines, but keeping the rest of those models' settings. This would be a lot easier to do by straight forward copy/paste of tabs rather than starting with a copy/paste model and then rebuilding the rest of settings manually, e.g. models typically differ from each other in thumbsnail image, receiver numbers, name sound files etc...

projectkk2glider, 2016-12-31:
Similar issue is #2718

kilrah, 2016-12-31:
> I've met many use cases where I wanted to extend a cluster of models by one feature, requiring some LS / SF defines, but keeping the rest of those models' settings.

This seems like a counter example... say you have to add 3 lines for the new function, wouldn't it make more sense to just copy those 3 manually than overwriting the whole tab and potentially having to redo everything that was different from the other model?

closed 2019-11-15 by stale[bot]