opentx/opentx #4179: translate companion to chinese, need a companion_zh_CN.ts file

Kongse, 2016-12-31:
Hi i'm trying to  to translate companion to chinese. 

At #712 I've seen some guide to do this, but seems  a companion_zh_CN.ts file should be created first. 

I'm new about github, so what should I do next.According to the issue workflow on wiki,seems I need fork the next branch, named "kongse/issueName_Number" and start to translate then?

thanks a lot.

projectkk2glider, 2016-12-31:
Is it possible to display Chinese texts using western charactes? If not (which I think is true), then this will be almost impossible. Current OpenTX code only has western fonts defined. Current characters can be seen here:

https://raw.githubusercontent.com/opentx/opentx/next/radio/src/fonts/480x272/font_midsize.png

kilrah, 2016-12-31:
He seems to be interested in translating Companion only, and posted on the forum that it works (he inserted Chinese translations in the italian file and they seem to display).

projectkk2glider, 2016-12-31:
:blush: Missed that.

Kongse, 2017-01-01:
Yes the companion can display Chinese but firmware seems impossible. 

It  comes a problem that when Chinsese users use a translated companion software to edit models, they may feel unsure what items on Radio corresponding to the items on companion. 

It can be partly solved by while translating the companion to Chinese, showing English words in brackets beside the translation words. I've tested it before (as kilrah mentioned above using italian ts file) and seems working well. 

So when they edit one item on companion, they know what's the item exactly corresponding to on the radio. And by cross reference, a chinese user unknown about English would learn to user Opentx radios faster.

kilrah, 2017-01-01:
There you go:
https://github.com/opentx/opentx/tree/next/companion/src/translations

Kongse, 2017-01-01:
Thank you all too :), I'm working on the file.

kilrah, 2017-01-08:
Translation included but somehow not working, still need to fix

kilrah, 2017-01-08:
Fixed

commit eb4508e1 (2017-01-01): [companion] Add zh_CN Chinese translation (fix #4179)
Fix lupdate working dir
commit e7f258ee (2017-01-08): Fix Chinese translation (fix #4179)
Remove disabled languages from build
closed 2017-01-08 by kilrah